adafruit/Adafruit_CircuitPython_NeoPixel #1: Support different pixel orders and RGBW strips

tannewt, 2016-11-30:
Right now the neopixel modules supports GRB order 400khz and 800khz pixel strips. We should update the module to support a wider range of pixel types:

RGB and other arbitrary byte orders
RGBW strips
This can probably all be done in the Python module as the low level pixel write function just takes an array of bytes to send out to the pixels (so it can change from 3-tuples to 4-tuples of color info for RGBW easily).

mcauser, 2016-12-05:
https://github.com/micropython/micropython/issues/2478
and
https://github.com/micropython/micropython/pull/2536

tannewt, 2018-01-24:
This is done. The constructor takes `bpp` which can be 4 for RGBW pixels.

caternuson, 2018-02-21:
Reopening to address the pixel order part of the issue. How about this:
- Create a dictionary to map string based pixel order definitions to associated order tuple
- Add parameter to `__init__` to allow passing this in
- Set the ORDER tuple from the dictionary using the passed in parameter as key

Ex:
`pixels = neopixel.NeoPixel(board.NEOPIXEL, 1, pixel_order="RGB")`

ladyada, 2018-02-21:
im cool w/this but rather than passing in "RGB" lets have a constant such as neopixel.RGB?

caternuson, 2018-02-21:
Additionally, `bpp` could be set based on this argument. But probably shouldn't remove `bpp` from `__init__`? (would possibly break current examples)

I'm fine with constants. So this:
Ex:
`pixels = neopixel.NeoPixel(board.NEOPIXEL, 1, pixel_order=neopixel.RGB)`

ladyada, 2018-02-21:
id leave `bpp` but prioritize if `order` is defined

caternuson, 2018-02-21:
There's a rough draft PR. Comment on it over there. Will make lint happy and update doc examples if this seems like the right approach.

ladyada, 2018-02-21:
i think we can just have the RGBW = (0, 1, 2, 3) rather than index?

caternuson, 2018-02-21:
Good call, was over thinking it. Removed the dict.

caternuson, 2018-02-24:
Closing back up. Different pixel order support added by: #24

closed 2018-02-24 by caternuson