adafruit/Adafruit_CircuitPython_NeoPixel #15: import of neopixel_write fails on ESP8266

jerryneedell, 2017-10-14:
neopixel.py uses
```
from neopixel_write import newpixel_write
```
but on the ESP8266 the "builtin" neopixel module uses:
```
from esp import neopixel_write
```
If the neopixel.py (or.mpy) library is loaded on an ESP8266 then 
``` import neopixel```
fails because neopixel_write cannot be found.

Since neopixel is "built-in" to the ESP8266 - this should not occur, but it will if the library is loaded.

dobra-dobra, 2017-10-15:
As it seems that's not the real problem. ESP8266 port of CP includes older versions of neopixel and neopixel_write modules, which are incompatible with current versions for SAMD ports.

jerryneedell, 2017-10-15:
@dobra-dobra - You are correct - the current ESP8266 CP implementation is not compatible with the "built-in" me-pixel module and the CP Bundle neopixel Module is not compatible with the ESP8266 neopixel_write.

It does appear that you can use neopixels on the ESP8266 as described fro micropython: https://docs.micropython.org/en/v1.8/esp8266/esp8266/tutorial/neopixel.html

The following wrks with a 7 neopixel string on GPIO15
```
import machine,neopixel
np=neopixel.NeoPixel(machine.Pin(15),7)
np[0]=(10,0,0)
np.write()
```

tannewt, 2017-10-16:
I think the missing `neopixel_write` is an oversight on my part. It should be a simple change to mpconfigport.h to add it to the ESP8266.

dobra-dobra, 2017-10-16:
Please compare neopixel and neopixel_write modules which are included in ESP8266 version of CP with those in SAMD version, they differ (in ESP8266 CP they are more or less the same as in MicroPython). You can't use neopixel module which is in driver bundle with neopixel_write module built-in in ESP8266 version of CP, they are incompatible.

tannewt, 2017-10-16:
Yup! Will do! Thanks for the heads up.

jerryneedell, 2017-10-17:
This has been address via issue #339 in circuitpython. The implementation of neopixe;_write has been corrected and the frozen neopixel.py module removed. The ESP8266 now works with this module

closed 2017-10-17 by jerryneedell