adafruit/Adafruit_CircuitPython_NeoPixel #16: should fill be a property?

caternuson, 2017-11-02:
Just a suggestion. Discuss....
This:
```python
pixels.fill = 0xFFBAD2
```
instead of this:
```python
pixels.fill(0xFFBAD2)
```

ladyada, 2017-11-02:
i think since it does it for each pixel, im leaning towards function...its not a property its an action "to fill" :)

caternuson, 2017-11-02:
by "property", i was thinking a python @ property function, not just a class member variable

dhalbert, 2017-11-02:
I agree with @ladyada. It's not a property, because it does something. It doesn't make sense to get the value (`f = np.fill`), only to set it.

You might make a similar argument about setting `.brightness`, but that preserves the states of the individual pixels: it turns a notional brightness knob on the whole set of pixels.

caternuson, 2017-11-02:
sorry sorry. i meant an @ enabled.setter. (i use the term property too loosely)

tannewt, 2017-11-03:
@caternuson I agree with @dhalbert and @ladyada.

Remember that its a matter of what it looks like from the outside. It doesn't matter that its a function internally. From the outside it should make sense as a variable rather than a function. Properties just make it convenient to store variables in a different way than they appear. For example, `pixels` is a bytearray rather than a list of tuples. Another example is the adc where the value is actually "stored" on the device.

caternuson, 2017-11-03:
Thanks for the responses. I think that's enough to go ahead and close for now.

closed 2017-11-03 by caternuson