adafruit/Adafruit_CircuitPython_NeoPixel #20: Add class variables for basic colors

Sigafoos, 2018-01-06:
I think it would be useful to have simple colors included as class variables. ie `neopixel.Red` or `neopixel.Color.Red` would be defined as `(0x10, 0, 0)`

tannewt, 2018-01-24:
@Sigafoos want to make a PR for this? Maybe it'd fit better in a different library so it could be used with other pixels like DotStars. Is there a Python lib we could copy?

kattni, 2018-01-24:
@tannewt and I had already discussed making a separate color/animations library. I agree this would be better as a separate library - less overhead when not being used, more opportunity for use with other libs. It was suggested that I add @PaintYourDragon to this conversation as well.

I think it would be great to start with just a color lib if you weren't interested in the animation concept, and then the lib is there if we want to add animations at a later time. 

If you were interested, the idea came up when discussing having the basic animations available in MakeCode setup in a library to be available for people coming from MakeCode to CircuitPython. They're called things like "running" and "theatre chase", it's easy to get an idea what they do by looking at MakeCode (there's a simulation available). As well you could add anything else you like, those were meant to be a starting point. Like I said though, this is a lot more involved than creating color variables, so it's still excellent to start with a color-only lib.

margaret, 2018-05-14:
I am going to take a stab at this (pycon sprints).

margaret, 2018-05-14:
Nvm, @apatt did this already

apatt, 2018-05-14:
I created the the library named led_animation and inside of that is color which has a list of named colors

tannewt, 2018-05-16:
Done in https://github.com/adafruit/Adafruit_CircuitPython_LED_Animation

closed 2018-05-16 by tannewt