adafruit/Adafruit_CircuitPython_NeoPixel #31: PyPi Install ImportError - no module named neopixel

brentru, 2018-08-24:
Hardware: Pi Zero W, latest blinka installation, Python 3.5.3

Python3: 
>  import neopixel
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ImportError: No module named 'neopixel'

Install Log:
> pi@io-pi:~ $ pip3 install adafruit-circuitpython-neopixel
Collecting adafruit-circuitpython-neopixel
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-neopixel/adafruit_circuitpython_neopixel-3.3.0-py3-none-any.whl
Collecting Adafruit-Blinka (from adafruit-circuitpython-neopixel)
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-0.2.3-py3-none-any.whl
Collecting Adafruit-GPIO (from Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Using cached https://www.piwheels.org/simple/adafruit-gpio/Adafruit_GPIO-1.0.3-py3-none-any.whl
Collecting spidev (from Adafruit-GPIO->Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.2-cp35-cp35m-linux_armv6l.whl
Collecting adafruit-pureio (from Adafruit-GPIO->Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-0.2.3-py3-none-any.whl
Installing collected packages: spidev, adafruit-pureio, Adafruit-GPIO, Adafruit-Blinka, adafruit-circuitpython-neopixel
Successfully installed Adafruit-Blinka-0.2.3 Adafruit-GPIO-1.0.3 adafruit-circuitpython-neopixel-3.3.0 adafruit-pureio-0.2.3 spidev-3.2

ladyada, 2018-08-24:
ohh i wonder if this is the line thats wrong? it should be just 'neopixel'
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/setup.py#L59
i think? @brentru wanna try fixing, re-releasing and see if that helps?

brentru, 2018-08-25:
Looks good, I can install it on Pi!

closed 2018-08-25 by ladyada