adafruit/Adafruit_CircuitPython_NeoPixel #34: Silently accepting mismatched color tuples.

caternuson, 2018-10-21:
Should we be silently accepting color tuples that don't match pixel order?
Give an RGB a 4 tuple, no errors:
```python
Adafruit CircuitPython 3.0.3 on 2018-10-10; Adafruit Gemma M0 with samd21e18
>>> import board, neopixel
>>> pixels = neopixel.NeoPixel(board.A2, 8, pixel_order=neopixel.RGB)
>>> pixels[0] = (255,0,0,255)
>>> pixels.fill((255,0,0,255))
```
Give a RGBW a 3 tuple, no errors:
```python
Adafruit CircuitPython 3.0.3 on 2018-10-10; Adafruit Gemma M0 with samd21e18
>>> import board, neopixel
>>> pixels = neopixel.NeoPixel(board.A2, 8, pixel_order=neopixel.RGBW)
>>> pixels[0] = (255,0,0)
>>> pixels.fill((255,0,0))
```
In both cases, all zeros get sent out instead. See here:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L145

kattni, 2018-10-21:
No, we should not. Seems like having that be caught and return an error would be super helpful.

caternuson, 2018-10-29:
The possible combinations can be summarized as:

| Input  | RGB NeoPixels | RGBW NeoPixels |
| :---: | :---: | :---: |
| 24 bit int  | OK  | automagic  |
| 32 bit int  | ?  | ?  |
| 3 tuple  | OK  | ?  |
| 4 tuple  | ?  | OK  |

We need to decide what the behavior should be for the `?`. The `OK` entries have obvious behavior. Nothing is currently done for `32 bit int`, adding it here for completeness.

jerryneedell, 2018-10-29:
its a bit hard to distinguish a 32 bit int from a 24 bit int with the 8 upper bit 0.
0x00123456
vs 
0x123456

jerryneedell, 2018-10-29:
so I guess you just wan tot reject a value with the upper nibble set for an RGB
that is
0xxx123456
will be reject if xx non zero for the RGB case.
Is that what you have in mind?

jerryneedell, 2018-10-29:
never mind -- clearly I'm confused about how the ints are used.
apparently the upper 8 bits are never used. is that correct?
Should they be used for an RGBW to set W ?

jerryneedell, 2018-10-29:
Discussed in weekly -- plan forward:
Throw ValueError if 4-tuple fed to RGB
Throw ValueError if any bits above bit 23 are set in integer mode

If 3-tuple sent to RGBW -- set RGB values.

caternuson, 2018-10-29:
Yep. In table form:

| Input  | RGB NeoPixels | RGBW NeoPixels |
| :---: | :---: | :---: |
| 24 bit int  | OK  | automagic  |
| >24bit int  | ValueError  | ValueError  |
| 3 tuple  | OK  | W=0  |
| 4 tuple  | ValueError  | OK  |

jerryneedell, 2018-10-29:
Thanks! -- hopefully this is accomplished but #37  Let me know if I missed something.

caternuson, 2018-10-30:
Closing. Fixed by #37. Thanks @jerryneedell.

Possible future solution - make things super simple here, just support a 3 or 4 tuple. Move all the logic and other things to some external helper that can take various inputs/flags/etc and return the required 3 or 4 tuple.

closed 2018-10-30 by caternuson