adafruit/Adafruit_CircuitPython_NeoPixel #41: NeoPixel with PCA9685 and adafruit_pca9685 module

JulianS-Uni, 2019-02-11:
Wouldn't it be nice to use NeoPixel with PCA9685 PWM extension boards? The problem is, that i have to give the NeoPixel an Pin but when using a PCA9685 PWM board and adafruit_pca9685 one does not have access to individual Pins, but to PWMChannels which are based on pulseio.PWMOut API.
Wouldn't it be a good approach to use something as the pulsio objects instead of the Pins in the NeoPixel module?

ladyada, 2019-02-11:
hiya - you cannot - its a totally different protocol!

closed 2019-02-11 by ladyada

tannewt, 2019-02-11:
Even if you treated each PCA pin as a DigitalInOut, the overhead of the I2C transaction would likely be too slow for the neopixel timing.

JulianS-Uni, 2019-02-12:
Hi thank you for your answer. I looked a little more into the NeoPixel protocol and i think i understood, that it is not a real PWM signal but a tight clocked digital signal. 
> Even if you treated each PCA pin as a DigitalInOut
Is this even possible?
> he overhead of the I2C transaction would likely be too slow for the neopixel timing
Does this mean it wouldn't be even possible to control a single LED or it would only be possible to control a limited amount of LEDs?