adafruit/Adafruit_CircuitPython_NeoPixel #42: Board not supported - beagle bone

CharlieBouyaka, 2019-02-25:
Hi,

I followed this tutorial : 
([](https://learn.adafruit.com/adafruit-neopixel-uberguide/python-circuitpython))

The set up was successful also with adafruit-blinka and everything we can find here : 
[](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi)

`debian@beaglebone:~$ sudo python3
[sudo] password for debian: 
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> 
>>> 
>>> 
>>> import board
>>> import neopixel
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/neopixel.py", line 34, in <module>
    from neopixel_write import neopixel_write
  File "/usr/local/lib/python3.5/dist-packages/neopixel_write.py", line 19, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
>>> `

Debian version (Debian 9.5 2018-10-07 4GB SD IoT  from beagleboard website): 

`debian@beaglebone:~$ cat /etc/*-release
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"`

All the best

ladyada, 2019-02-25:
not surprising, i dont know we ever got neopixels working on beaglebone via Blinka! right now only Raspberry Pi linux computers have support

CharlieBouyaka, 2019-02-25:
That's a bad news... I like my beaglebone...
I'm trying to use a neopixel led strip with...
Have you planned to come to beaglebone ?

ladyada, 2019-02-25:
nope no plans!

CharlieBouyaka, 2019-02-26:
ah! Thank you for the feedback!
All the best, :)

giuliomoro, 2019-09-03:
@CharlieBouyaka did you try some of the BeaglBone + PRU neopixels libraries? e.g.: 
https://github.com/toroidal-code/PixelBone (note: I didn't try it!)