adafruit/Adafruit_CircuitPython_NeoPixel #43: ImportError: No module named board

karishmabava, 2019-03-18:
Hi,

i am running it on Rpi yet it is giving me this error. Running the command as root.

Please help1

Karishma

ladyada, 2019-03-19:
hiya please follow this guide
https://learn.adafruit.com/neopixels-on-raspberry-pi

closed 2019-03-19 by ladyada

eddiewebb, 2020-01-28:
@karishmabava after folloing the guide linked it failed for me, i found a force reinstall of blinka worked.

`sudo python3 -m pip install --force-reinstall adafruit-blinka`

seyo-IV, 2020-03-02:
Having the same issue, force reinstall didn't help. Still missing the module.

caternuson, 2020-03-02:
When you ran the Blinka install check, it ran without errors?
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3-15

devin-perry, 2020-03-20:
@caternuson I ran `python3 blinkatest.py` and got this error:
`Traceback (most recent call last):`
   `File "blinkatest.py", line 5, in <module>`
      `"execution_count": null,`
`NameError: name 'null' is not defined`
Any idea how to fix this?

devin-perry, 2020-03-20:
@caternuson I ran `python3 blinkatest.py` and got:

`File "blinkatest.py", line 1, in <module>`
`import board`
`ModuleNotFoundError: No module named 'board'`

gazhay, 2020-04-24:
I followed the tutorial - it worked - rebooted and now get no module errors for "board".
Literally only rebooted the pi.

Tried removing all and reinstalling, still didn't work.

mhaith, 2020-04-29:
Same issue, ”No module named board,” even after force reinstall of blinks and run sudo update and sudo upgrade.  Strangely, the python programs for the lsm9ds1 module I am using run perfectly when I bring them up in the idle editor and run them with the “run module,” but when I run them with the command line (which I need to start them from cron), I get this error message.

mhaith, 2020-04-29:
Problem solved.  Silly me.  I was stipulating Python in the CLI rather than Python3.  Needed to sleep on it.

avvy65, 2020-11-13:
I have been getting this error and did a force reinstall of adafruit blinka, ran the blinkatest.py script without error, but when I ran other adafruit python scripts I still got the same error. I used sudo python3 to run the scripts. As I don't know what else to do I've just given up with trying to use python scripts on a raspberry pi.

ben-kenney, 2021-01-09:
I ran into this issue as well. After running:

```
sudo python3 -m pip install --force-reinstall adafruit-blinka
```

then then ```import board``` worked for me.

The ```--force-reinstall``` instruction is in the [Python Installation of NeoPixel Library](https://learn.adafruit.com/neopixels-on-raspberry-pi/python-usage) but might need to also be added to the [Installing CircuitPython Libraries on Raspberry Pi](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3-15) as well.

danielfaust, 2021-10-06:
If you installed it into a non-sudo virtual environment or as a user package, and then start the script by issuing `sudo` (in order to get access to the hardware), then the modules won't be available to Python. The solution would be to add the user to the group which has access to the hardware (ie `i2c`) and run the script without `sudo`.

danielfaust, 2021-10-07:
@nishantbb `crontab` uses `/bin/sh` as the default shell, so maybe there is some initialization done in `.bashrc` or another location which is not getting executed. Maybe adding `SHELL=/bin/bash` to the top of the crontab file would help. That is, if it is the user's cron, else I think it runs as root, in which case this would not work, as long as you're not executing the script as the user which installed the libraries. In that case ["crontab running as a specific user"](https://serverfault.com/questions/352835/crontab-running-as-a-specific-user/352837#352837) could be helpful.