adafruit/Adafruit_CircuitPython_NeoPixel #46: Default pixel_order and bpp don't match

caternuson, 2019-05-22:
If NeoPixels are created without specifying a `bpp` or a `pixel_order`:
```python
pixels = neopixel.NeoPixel(board.D18, 8)
```
then this would get used:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/c0bdd8b10383725ee9293f5d88fb8d47eb1272bd/neopixel.py#L94-L96
and `bpp=3` by default:
```python
 def __init__(self, pin, n, *, bpp=3, brightness=1.0, auto_write=True, pixel_order=None):
```
so you end up with `bpp=3` and GRBW for `pixel_order`, which is `bpp=4`.

This is my bad, since it was introduced with #24 to fix #1. I think there was some attempt to maintain backwards compatibility with the older style of using `bpp` only. So that's why both `bpp` and `pixel_order` are still there:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/1#issuecomment-367432340

But might be good to somehow check and protect against this:
https://forums.adafruit.com/viewtopic.php?f=47&t=151980

ladyada, 2019-05-22:
oops yeah we are usually wordy in the definitions - we could warn people for now they must define the color order?

caternuson, 2019-05-22:
Just to throw out the suggestion - how about removing `bpp` from the parameter list? `pixel_order` is all that is really needed. This would maybe break some things, but would be cleaner. And then any older examples that are still using `bpp` should be updated to use `pixel_order`.

ladyada, 2019-05-22:
im ok with that, do any of our examples use bpp?

caternuson, 2019-05-22:
All the examples here are using `pixel_order`. Not sure what else might be out there.

@tannewt You were more involved in the `bpp` approach. Any reason to keep it?

tannewt, 2019-05-22:
I'd ask @rhooper since he's got some big changes in the pipe for the RGB led libs.

caternuson, 2019-06-05:
@rhooper - see above. What are the changes?

caternuson, 2019-08-21:
@tannewt Any idea of status of above w.r.t. "big changes"? I'm in favor of getting rid of `bpp` in favor of just using `pixel_order`.

rhooper, 2019-08-21:
The changes will be to use _pixelbuf which will be using a string to compute everything to do with the byte order and bpp.  Bpp will then be able to be based on that string.  For backwards compatibility we can check its a sane value and ignore otherwise.

caternuson, 2019-08-21:
@rhooper thanks. can you link to where this is being done? it's still not clear to me how this applies here.

rhooper, 2019-08-21:
Here's the repos for the WIP.  Ping me on Discord if you want to discuss in more real-time.  Dan also wants to sync up on this work soon.

https://github.com/rhooper/circuitpython/tree/new-pixelbuf-api
https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf
https://github.com/rhooper/Adafruit_CircuitPython_NeoPixel/tree/pixelbuf
https://github.com/rhooper/Adafruit_CircuitPython_DotStar/tree/pixelbuf

rhooper, 2020-01-14:
@caternuson is this still an issue?

caternuson, 2020-01-14:
@rhooper Thanks! Yah, this is probably fixed with the new logic. Let's close this for now. I think it might still be worth considering dumping `bpp` all together and use `pixel_order` for everything. But we can defer that to a new separate issue.

closed 2020-01-14 by caternuson

houshmandX, 2023-02-02:
Hi all,

I have recently purchased a [NeoPixel Ring - 12 x 5050 RGB LED from amazon](https://www.amazon.com/dp/B00KAE3R1U?psc=1&ref=ppx_yo2ov_dt_b_product_details).

I have also followed the link https://cdn-learn.adafruit.com/downloads/pdf/neopixels-on-raspberry-pi.pdf
to install NeoPixel Library. 

So, I can turn the LEDS, but they are not acting as desired, for instance, colors are randomly changing/blinking.

So, for the test, I want to turn all the LEDs on and be colored red:

`import time
import board
import neopixel

pixel_pin = board.D18
num_pix = 12
order = neopixel.GRBW # won't matter if I use orders, RGB, RGBW, GRB

pixel = neopixel.NeoPixel(pixel_pin, num_pix, brightness = 0.2, auto_write = False, pixel_order = order)

while True:
    pixel.fill((255,0,0))
    pixel.show()
`

what am I missing, why, few of LED gets change colors, blinking and etc.

caternuson, 2023-02-02:
@houshmandX Please post in the forums with photos of your setup:
https://forums.adafruit.com/