adafruit/Adafruit_CircuitPython_NeoPixel #47: Driver causes "Segmentation Fault" at end of python app.

jhandel, 2019-07-06:
I am getting an error when my Python app closes.. it is throwing a Segmentation Fault outside the app as its closing (after all of my python code has executed) when I create or interact with a neopixel from a separate thread.

Behavior: If a Neopixel is used or created inside a separate thread then after python has closed the app a "segmentation fault" happens

Expected: No Segmentation fault

Environment: 
     Pi 3
     Python 3.5.3
     CircuitPython -- Latest
     Neopixel driver -- Latest

Example Script:

```
# Simple test for NeoPixels on Raspberry Pi
import time
import board
import neopixel
import threading

# Choose an open pin connected to the Data In of the NeoPixel strip, i.e. board.D18
# NeoPixels must be connected to D10, D12, D18 or D21 to work.
pixel_pin = board.D18

# The number of NeoPixels
num_pixels = 30

# The order of the pixel colors - RGB or GRB. Some NeoPixels have red and green reversed!
# For RGBW NeoPixels, simply change the ORDER to RGBW or GRBW.
ORDER = neopixel.GRB

pixels = neopixel.NeoPixel(pixel_pin, num_pixels, brightness=0.2, auto_write=False,
                           pixel_order=ORDER)
play = True

def wheel(pos):
    # Input a value 0 to 255 to get a color value.
    # The colours are a transition r - g - b - back to r.
    if pos < 0 or pos > 255:
        r = g = b = 0
    elif pos < 85:
        r = int(pos * 3)
        g = int(255 - pos*3)
        b = 0
    elif pos < 170:
        pos -= 85
        r = int(255 - pos*3)
        g = 0
        b = int(pos*3)
    else:
        pos -= 170
        r = 0
        g = int(pos*3)
        b = int(255 - pos*3)
    return (r, g, b) if ORDER == neopixel.RGB or ORDER == neopixel.GRB else (r, g, b, 0)

def rainbow_cycle(wait):
    for j in range(255):
        for i in range(num_pixels):
            pixel_index = (i * 256 // num_pixels) + j
            pixels[i] = wheel(pixel_index & 255)
        pixels.show()
        time.sleep(wait)

def pixel_play():
    while play:
        # Comment this line out if you have RGBW/GRBW NeoPixels
        pixels.fill((255, 0, 0))
        # Uncomment this line if you have RGBW/GRBW NeoPixels
        # pixels.fill((255, 0, 0, 0))
        pixels.show()
        time.sleep(1)

        # Comment this line out if you have RGBW/GRBW NeoPixels
        pixels.fill((0, 255, 0))
        # Uncomment this line if you have RGBW/GRBW NeoPixels
        # pixels.fill((0, 255, 0, 0))
        pixels.show()
        time.sleep(1)

        # Comment this line out if you have RGBW/GRBW NeoPixels
        pixels.fill((0, 0, 255))
        # Uncomment this line if you have RGBW/GRBW NeoPixels
        # pixels.fill((0, 0, 255, 0))
        pixels.show()
        time.sleep(1)

        rainbow_cycle(0.001)    # rainbow cycle with 1ms delay per step

thread = threading.Thread(target=pixel_play)
thread.start()
time.sleep(10)
play = False
thread.join()
print("thread completed")
print("app completed")
```

Output:
```
sudo python neopixel_threaded.py
thread completed
app completed
Segmentation fault
```

ladyada, 2019-07-06:
that's normal, and comes from the rpi_ws281x library we use. not much we can do about it (its probably a side effect from the 'unique' and 'exciting' way we have to trick the pi into driving neopixesl!)

jhandel, 2019-07-06:
Any reason I can't keep running them in threads if I am ok with the unfortunate error message at the end? (working on a simon says for fencers and want to run each neopixel ring in its own thread so I can issue animations to them separate from listening for input)... Don't want to find out down the road that this was a bad way to handle 5+ animations + 5+ analog to digital signals all being processed for the game after I was deeper than P.O.C. into the code.

ladyada, 2019-07-06:
oi - it may not be compatible... neopixels on raspi are very finicky like that. i would have one thread that does not quit, and you can send it messages on what to display. but dont uninit the neopix!

dhocker, 2019-09-16:
I believe the problem you are seeing is caused by a memory leak that manifests itself when your app closes. I have an app that is similar to yours and I have seen the segmentation fault message and a memory leak message. The memory leak message seems to occur most of the time.

The problem is that the NeoPixel class fails to clean up its use of the rpi_ws281x module when it is destroyed. The problem is compounded by the fact that the function needed to properly clean up is actually deep in Adafruit_Blinka and is not exposed by Adafruit_Blinka. You can find it here: Adafruit_Blinka/src/adafruit_blinka/microcontroller/bcm283x/neopixel.py (look for neopixel_cleanup).

I have forked the Adafruit_Blinka and Adafruit_CirecuitPython_NexPixel repos and attempted to fix the problem. You can obtain the forked repos here:

https://github.com/dhocker/Adafruit_Blinka.git
https://github.com/dhocker/Adafruit_CircuitPython_NeoPixel.git

WRT threading, as far as I can tell, none of the code in NeoPixel, Blinka or rpi_ws281x appears to be specifically thread safe. There are no semaphores or locks guarding code that is serially reusable. As long as you are using the NeoPixel instance on one and only one thread you are probably safe (that's how my app works). If you need to use the NeoPixel instance on multiple threads you'll probably need to implement your own locking so that only one thread can be using the NeoPixel instance at any given time.

If you want to try this fix, you can uninstall the stock Adafruit_Blinka and Adafruit_CircuitPython_NeoPixel packages using "pip uninstall". Install the forked versions by running "python setup.py install" for each repo.

I hope this helps you out.

dhocker, 2019-09-18:
@jhandel, I believe the problem you see is caused by clean up issues in Adafruit_Blinka and Adafruit_CircuitPython_NeoPixel. I have seen two distinctly different issues. One case is a memory leak reported at app exit. The other case is like yours, namely a segment fault reported after your app exits.

I have forked the Adafruit_Blinka and Adafruit_CirecuitPython_NexPixel repos and attempted to fix both problems. You can obtain the forked repos here:

https://github.com/dhocker/Adafruit_Blinka.git
https://github.com/dhocker/Adafruit_CircuitPython_NeoPixel.git

The memory leak seems to be caused by clean up code in NeoPixel. The cleanup code seems to run after the memory leak is reported when your app closes. The segment fault seems to be caused by missing cleanup code in the Pin class implementation for the RPi.

The changes I made in the forked repos attempt to fix both of these issues.

WRT threading, as far as I can tell, none of the python code in NeoPixel, Blinka or rpi_ws281x appears to be specifically thread safe. There are no semaphores or locks guarding code that is serially reusable. As long as you are using the NeoPixel instance on one and only one thread at a time you are probably safe. If you need to use the NeoPixel instance on multiple threads you'll probably need to implement your own locking so that only one thread can be using the NeoPixel instance at any given time. In my app, I create an instance of the NeoPixel class on the main thread and then use the instance on a spawned thread. This seems to work fine.

Also, it appears that the NeoPixel class is intended to be created once and only once per app run. If you attempt to create an instance of NeoPixel, use it and then delete it, I believe you will find that you really can't create another instance. Similarly, NeoPixel has support for the "with" statement. But, at the end of the "with" statement the NeoPixel instance is effectively destroyed (it is no longer useable).

If you want to try my fixes, you can uninstall the stock Adafruit_Blinka and Adafruit_CircuitPython_NeoPixel packages using "pip uninstall". Install the forked versions by running "python setup.py install" for each repo.

I hope this helps you out.