adafruit/Adafruit_CircuitPython_NeoPixel #50: Allow 32 bit color spec for RGBW NeoPixels?

caternuson, 2019-08-21:
Suggested format would be `0xWWRRGGBB` to match Arduino lib:
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.h#L312

Currently, only way to control W channel directly is via tuple:
```python
>>> pixels.fill((0,0,0,255))
```
this isn't allowed:
```python
>>> pixels.fill(0xFF000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "neopixel.py", line 212, in fill
  File "neopixel.py", line 172, in __setitem__
  File "neopixel.py", line 135, in _set_item
ValueError: only bits 0->23 valid for integer input
>>> 
```

caternuson, 2019-08-21:
Seesaw also allows it:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/master/adafruit_seesaw/neopixel.py#L90
Format there is also `0xWWRRGGBB`.

tannewt, 2019-08-22:
I'm not sure I like this because the fourth byte tends to be alpha in graphics: https://en.wikipedia.org/wiki/RGBA_color_space

caternuson, 2019-09-04:
Hmmmm. OK, good point. Anyway you think this could be done? Or do you think individual RGBW color channel control should be limited to specifying via a tuple?

tannewt, 2019-09-04:
Ya, I would only support RRGGBB and not worry about white. Would it help to disable int value support for RGBW pixels?

caternuson, 2019-12-04:
Nah, I don't think so. That might just add a different kind of confusion. Essentially this library does not currently support 4 byte color values. The Arduino and Seesaw libs have decided to make 4 byte values be WRGB. So if this library adds support for 4 byte values, will have to decide then if it should be different or not.

I'm going to close this for now. Nothing is currently broken and control of the W channel is possible via tuple. Can always reopen if this becomes important again.

closed 2019-12-04 by caternuson