adafruit/Adafruit_CircuitPython_NeoPixel #51: Board not supported  (Jetson Nano)

cscholler, 2019-08-22:
Greetings,

I am trying to follow this article (http://blog.whatgeek.com.pt/2015/04/raspberry-pi-and-adafruits-neopixel-stick/) but running into "Board not supported". I am using the Jetson Nano.
Here is the error below:

Exiting... 
Cleaning up pins
Traceback (most recent call last):
  File "/home/jetson/PycharmProjects/servoTest/venv/ledTest.py", line 3, in <module>
    from neopixel import *
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/neopixel.py", line 34, in <module>
    from neopixel_write import neopixel_write
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/neopixel_write.py", line 19, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported

Process finished with exit code 1

Here is a list of my packages:

Adafruit-Blinka | 2.3.1 | 2.3.1
-- | -- | --
Adafruit-GPIO | 1.0.3 | 1.0.3
Adafruit-PlatformDetect | 1.3.2 | 1.3.2
Adafruit-PureIO | 0.2.3 | 0.2.3
Jetson.GPIO | 2.0.0 | 2.0.0
RPi.GPIO | 0.7.0 | 0.7.0
adafruit-circuitpython-busdevice | 3.0.0 | 3.0.0
adafruit-circuitpython-motor | 2.0.0 | 2.0.0
adafruit-circuitpython-neopixel | 3.3.7 | 3.3.7
adafruit-circuitpython-pca9685 | 3.2.5 | 3.2.5
adafruit-circuitpython-register | 1.7.1 | 1.7.1
adafruit-circuitpython-servokit | 1.1.0 | 1.1.0
board | 0.0.0.post0 | 0.0.0.post0

ladyada, 2019-08-22:
you can subscribe to this issue here for the board not detected issue
https://github.com/adafruit/Adafruit_Blinka/issues/141

note that neopixels are not supported on the jetson nano natively at this time, please ask the jetson nano developers to do so and we'll add the interface support!

closed 2019-08-22 by ladyada

cscholler, 2019-08-23:
@ladyada will do.

cscholler, 2019-08-23:
Someone have already requested it.
https://devtalk.nvidia.com/default/topic/1061833/jetson-nano/support-for-neopixels-adfruit/