adafruit/Adafruit_CircuitPython_NeoPixel #52: list rgb value reported to be slower than tuple

dhalbert, 2019-09-26:
FlakShim in discord #help-with-circuitpython on Sept 25 reports that `pixels[i] = (10,10,10)` is 4x slower than `pixels[i] = [10,10,10]`. Confirm and investigate. No obvious issue noted in NeoPixel code; might be CircuitPython related.

rhooper, 2020-01-14:
This should be re-tested with _pixelbuf.

kattni, 2020-05-01:
@dhalbert Have you had a chance to test this with the recent updates to see if it is still an issue?

FoamyGuy, 2020-07-04:
I tested this on a CPX with v6.0.0-alpha.1 using this code:
```python
import time
import board
import neopixel
 
pixel_pin = board.NEOPIXEL
num_pixels = 10
 
pixels = neopixel.NeoPixel(pixel_pin, num_pixels, brightness=0.3, auto_write=True)

before_time = time.monotonic_ns()
pixels[0] = (10,10,10)
after_time = time.monotonic_ns()
print("tuple took: {}ns".format(after_time - before_time))

before_time = time.monotonic_ns()
pixels[0] = [10,10,10]
after_time = time.monotonic_ns()
print("list took: {}ns".format(after_time - before_time))
```

As far as I can tell there is not a major difference in the time it takes to set a tuple vs. setting a list. The does seem consistently a bit faster, but only slightly so, not multiple times faster. 

Here is an example of the output from this test:
```
tuple took: 2258305ns
list took: 2197269ns
```

I've run it several times the specific values differ slightly from run to run, but the overall theme remains the same tuple is slightly slower than list, but not slower by multiple times.

FoamyGuy, 2020-07-04:
Maybe I spoke too soon, or else there is a little bit more at play in the context of this issue... 

I just ran essentially the same test on an Edge Badge with v6.0.0-alpha.1 and the latest neopixel.mpy out of the current bundle. 

With that setup I am seeing a decently sized difference in the time it took to set the neopixel. However in this case the list seems to be slower than the tuple: 

```
tuple took: 244144ns
list took: 3051760ns
```

tekktrik, 2024-03-28:
Poking around at issues, I'm curious if this is still the case in 9.0.0.