adafruit/Adafruit_CircuitPython_NeoPixel #54: Can't get this to work with Raspi 4 or Raspi Zero on Buster

spiff72, 2019-10-05:
I get strange outputs - mostly full intensity white colors from the Neopixels when using this library on both of my Raspis with Buster installed.

I have tried both the neopixel_pixel example and the neopixel_rpi_simpletest example with similar results - outputs are dominated by blinding white light, with what appear to be flashes of normal behavior.

I have also tried with both an 8-pixel RGBW strip and single RGB pixels.  I have adjusted the pixel order and pixel counts to suit both situations with no success.  I don't have another pi with stretch (or earlier) installed to try and see if buster is really the problem, and it is late here (and time to give up for the night!)...

I am also using pin 10 as the data pin, since all the other options (12, 18, 21) require running as root.  Is this my problem?  If i run the python program as sudo, it gives an error stating that the 'import board' line fails (no such module - or something like that).

spiff72, 2019-10-05:
Well - something just isn't right here.  I just dug out another PiZero running stretch, installed circuitpython and this neopixel library and I get the same result.  I am stumped at this point.

jerryneedell, 2019-10-05:
How many neopixels are you driving and how are you powering them?
I tried with 16 neopixels (neopixel ring) just powered from the RPi 3.3 and the neopixel_rpi_simpetest.py worked on pin D18 but: 
I had to reinstall blinka and the neopixel lib as root
```
sudo pip3 install adafruit-blinka
sudo pip3 install adafruit-circuitpython-neopixel
sudo python3 neopixel-rpi-simpletest.py
```

spiff72, 2019-10-05:
Sorry - I should have posted, but I just got it working here.  I did some more research into why running the python script as root was giving the error about not finding modules, and found that using the -E switch with sudo allows for the environment to "stick" from normal usage.  This allowed the script to find the modules needed and it works.

However, I did have to switch to D21 for the data pin.  Even with root it doesn't work with pin 10.  I can't use pin D18 because that pin is used on the fanshim board I have installed on my Pi 4 for fan control.

I suspect that you running the installers as root installed them for the root user, and then they were visible when running the script with sudo.  Running the following works without the need for this (based on my research this afternoon):
`sudo -E python3 neopixel-rpi-simpletest.py`

(And making the adjustments to the pin).

I just re-checked here to see if running with the -E option and sudo on pin D10 works, and it still gets random flashes of bright white.  This no longer happens at all with pin D21.

spiff72, 2019-10-05:
I should also add that I poked around in the WS281x library and learned more about the pins - that was what got me thinking that the sudo operation was key, and to figure out HOW to run it with sudo.

closed 2019-10-05 by spiff72