adafruit/Adafruit_CircuitPython_NeoPixel #55: 'module' object has no attribute 'D18'

francescofact, 2019-10-12:
Hello, on my raspberry pi i get this error related to "board.D18".

The code line is:
pin = digitalio.DigitalInOut(board.D18)

The error is:
AttributeError: 'module' object has no attribute 'D18'

I have all the package installed and upgraded at the latest version.

jerryneedell, 2019-10-12:
did you install "blinka" as in this guide ? https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

also did you install a module called "board"  -- you should not have -- board is included in "adafruit_blinka"

jerryneedell, 2019-10-12:
Also  -- takes look at this guide https://learn.adafruit.com/neopixels-on-raspberry-pi/overview

francescofact, 2019-10-12:
thank you. 
i uninstalled board with pip3 (i don't know what package it is. it was already installed) and reinstalled "adafruit_blinka". now everything works. thank you

closed 2019-10-12 by francescofact

paulesbats, 2024-05-03:
shhhtt