adafruit/Adafruit_CircuitPython_NeoPixel #67: 'NeoPixel' object has no attribute 'buf'

hwende, 2020-01-18:
I just set up a new Raspberry 3.
I did   `pip3 install adafruit-circuitpython-neopixel`
This installed version 4.1.0

No when I start any program this happens:
```
File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 260, in __setitem__
    self.show()
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 151, in show
    neopixel_write(self.pin, self.buf)
AttributeError: 'NeoPixel' object has no attribute 'buf'
```

When I do this:    `pip3 install adafruit-circuitpython-neopixel==4.0.0`
everything works again.

I hope this helps.
Sincerely HWende

ladyada, 2020-01-18:
hi please try uninstalling adafruit-circuitpython-neopixel and adafruit-circuitpython-pypixelbuf and reinstalling both :)
https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf/releases/tag/1.0.1

closed 2020-01-19 by ladyada

dustyyeti, 2020-01-20:
If I sudo pip3 install adafruit-circuitpython-neopixel, I can run a python neopixel code with sudo
If I pip3 install (without sudo) I get the following error (when running the code with general user, no sudo)

>  python3 foo.py
Failed to create mailbox device
: Operation not permitted
Traceback (most recent call last):
  File "foo.py", line 7, in <module>
    pixels[0] = (255, 0, 0)
  File "/home/caps/.local/lib/python3.7/site-packages/adafruit_pypixelbuf.py", line 265, in __setitem__
    self.show()
  File "/home/caps/.local/lib/python3.7/site-packages/neopixel.py", line 151, in show
    neopixel_write(self.pin, self.buf)
  File "/home/caps/.local/lib/python3.7/site-packages/neopixel_write.py", line 24, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
  File "/home/caps/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 65, in neopixel_write
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.

ladyada, 2020-01-20:
you need to be root because the neopixel code underneath is doing really delicate memory poking that user permissions dont have!

dustyyeti, 2020-01-22:
Well that's the first time that answer has appeared WRT to the adafruit_neopixel code!

I've been chasing wild geese on forums for several days, in attempt to track down a reliable method to grant ordinary user access to the GPIO pins. Which I **seem to have** already, it's only the python scripts using imported libraries from adafruit code that will not behave in this manner.

I'm no hardware savant; can you comment as to why a raspberry pi developer can get his GPIO pins to respond to an ordinary user, but this neopixel routine is so delicate? Maybe I'm conflating the issues--the delicacy you are speaking of has nothing to do with GPIO, but merely the processes the code is activating.. (in memory...?)

This work is for a research lab, and my only goal is to have simple bash script control of the neopxiel, among other things. At this time, the current version of my code utilizes an Arduino accessory as the light controller, and this it does flawlessly without superuser access. I've been looking forward to optimizing the procedure and eliminating the extra hardware, since the RPi has I/O that should be able to handle something so simple. 

Thanks for your thoughts! It will be more satisfying to understand why my goals are out of reach, than to simply give up.

ladyada, 2020-01-22:
GPIO access is permitted without root access, mmio is not. you have to run any and all neopixel code on a Pi as root

dustyyeti, 2020-01-22:
Thank you for your response.