adafruit/Adafruit_CircuitPython_NeoPixel #69: neopixel library can fail to update NeoPixels after exceptions due to auto_write changes

kevinjwalters, 2020-01-19:
This may be less relevant now given current work going on in this library but the library I've got from a recent bundle can leave the cp.pixels object damaged after an exception from `fill()`:

```
Adafruit CircuitPython 5.0.0-beta.3 on 2020-01-08; Adafruit Circuit Playground Bluefruit with nRF52840
>>> from adafruit_circuitplayground import cp
>>> cp.pixels.fill(0)
>>> cp.pixels.fill(140737488355328)  ### erroneous value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "neopixel.py", line 212, in fill
  File "neopixel.py", line 172, in __setitem__
  File "neopixel.py", line 135, in _set_item
ValueError: only bits 0->23 valid for integer input
>>> cp.pixels.fill(1)
>>> cp.pixels.fill(256)
>>> cp.pixels.fill((10, 10, 10)) # last three fill() did not work
```

A quick look at the code (e.g. https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/64f091e96229b290e0cafc951fc9c626aae698f9/neopixel.py#L207-L215) reveals that `self.auto_write` is being changed in `fill` and not restored if there's an exception.

ladyada, 2020-01-19:
sure, wanna submit a PR that puts the `for` loop into a try/except/finally?

kevinjwalters, 2020-01-19:
I haven't followed the `PixelBuf` story that I believe @rhooper has been working on recently but hasn't this logic almost all gone into C code now? I see this https://github.com/adafruit/circuitpython/blob/master/shared-bindings/_pixelbuf/__init__.c#L91-L105

If that's about to ship then does it make sense to go back and change the python verison? It would require a separate branch to maintain that as the `PixelBuf` version is already on master.

ladyada, 2020-01-19:
yes it makes sense, linux/desktop CPython will use this library

kevinjwalters, 2020-01-21:
@tannewt What's the git approach here with the linux (is this called Blinka?) version of this library vs the `PixelBuf` accelerated versions of it?

(There was a brief mention of modification to the change/approval policy for this library in today's meeting.)

tannewt, 2020-01-21:
Please put the try: except: inside the for loop to make it clear what is throwing the exception. Thanks!

kevinjwalters, 2020-01-21:
Which branch do you want this on? This code doesn't exist anymore on master?

tannewt, 2020-01-22:
It is here now: https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf/blob/master/adafruit_pypixelbuf.py#L322

kevinjwalters, 2020-01-22:
Thanks, I didn't know that existed.

I don't think I can do the admin perm to move this ticket? Do you want to shift it over to  https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf ?