adafruit/Adafruit_CircuitPython_NeoPixel #71: Pixel buffer accepts tuples but returns lists

eddiewebb, 2020-01-27:
given the code, it fails.

```
red = (255,0,0)
pixels[1] = red
print( pixels[1])
```

Prints `[40,0,0]`

eddiewebb, 2020-01-27:
Not sure why the list/tuple change. But change in value is from https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L232 which multiplies by brightness.

caternuson, 2020-01-27:
Can you post full code so we can also see how you initialized the NeoPixel object.

tannewt, 2020-01-27:
What version of the library are you using?

eddiewebb, 2020-01-27:
@caternuson, the code is in a public repo. You can see where I define colors
https://github.com/eddiewebb/sense-show/blob/master/led_strip.py#L22

And where I basically want to say "if the pixel is not already set to X, set it to Y". 
https://github.com/eddiewebb/sense-show/blob/master/led_strip.py#L133

@tannewt 4.1.0
```
sudo pip3 freeze | grep neopixel
adafruit-circuitpython-neopixel==4.1.0
```

caternuson, 2020-01-27:
I think you might be hitting this:
https://github.com/adafruit/Adafruit_CircuitPython_Pypixelbuf/pull/4#discussion_r368309515
which should be fixed in latest release. Try updating to the 5.0.0 release and see if the problem persists:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/releases/tag/5.0.0
Return may still be a list, but hopefully will not have the incorrect value.

kattni, 2020-05-04:
This issue is resolved.
```
>>> import board
>>> import neopixel
>>> pixels = neopixel.NeoPixel(board.NEOPIXEL, 10, auto_write=False, brightness=0.5)
>>> red = (80, 0, 0)
>>> pixels[1] = red
>>> print(pixels[1])
(80, 0, 0)
```

closed 2020-05-04 by kattni