adafruit/Adafruit_CircuitPython_NeoPixel #72: There is a way to use NeoPixels without root privileges ?

MkLHX, 2020-02-10:
Hi there, 

I'm try to use neopixels in an iot project, i can run my script with 

> sudo python3 myscript.py

But i need to run neopixels as a python package from my script like

script_1.py
`
import time
import board
import neopixel

# Choose an open pin connected to the Data In of the NeoPixel strip, i.e. board.D18
# NeoPixels must be connected to D10, D12, D18 or D21 to work.
pixel_pin = board.D18

# The number of NeoPixels
num_pixels = 9

# The order of the pixel colors - RGB or GRB. Some NeoPixels have red and green reversed!
# For RGBW NeoPixels, simply change the ORDER to RGBW or GRBW.
ORDER = neopixel.GRB

pixels = neopixel.NeoPixel(pixel_pin,
                           num_pixels,
                           brightness=0.2,
                           auto_write=True,
                           pixel_order=ORDER)

COLORS = {
    'red': (255, 0, 0),  # red
    'green': (0, 255, 0),  # green
    'blue': (0, 0, 255),  # blue
    'white': (255, 255, 255),  # white
    'black': (0, 0, 0),  # black
}

LED_PINS = {
    'device_name_1': 0,
    'device_name_2': 1,
    'device_name_3': 2,
    'device_name_4': 3,
    'by': 4,
    'gpk': 5,
    'logo': 6,
    'ec': 7,
    'ph': 8,
}

def ctrl_led(led_num, color):
    """
    @Param led num => [0 to 8]
    @Param color list like (0, 255, 0)
    """
    pixels[led_num] = COLORS['black']
    time.sleep(0.125)
    pixels[led_num] = color
`
script_2.py
`
from script_1 import ctrl_led, LED_PINS, COLORS

if False is pump_is_running:
    ctrl_led(LED_PINS['ec'], COLORS['red'])
else:
    ctrl_led(LED_PINS['ec'], COLORS['blue'])
`

caternuson, 2020-02-10:
Run script_2.py with sudo? You'll need root one way or the other:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/67#issuecomment-576454207

MkLHX, 2020-02-10:
Ok can't pass over root privilege. 
So if i understand, i have to make a script.py and call it from python script by using subprocess.checkt_output() package or os.system() by passing sudo in these tools

caternuson, 2020-02-10:
Easiest is to just have everything in script.py and call it with sudo. Otherwise you'll need to make sure whatever is trying to make the actual NeoPixel access call has root privilege somehow.

MkLHX, 2020-02-13:
ok i manage by using my script like this:
`
from script_1 import COLORS, LED_PINS
if False is pump_is_running:
    subprocess.check_output(['sudo', 'python3', '-c', "import script_1; script_1.ctrl_led(%d, %s);" % (LED_PINS['ec'], COLORS['red'])])
else:
    subprocess.check_output(['sudo', 'python3', '-c', "import script_1; script_1.ctrl_led(%d, %s);" % (LED_PINS['ec'], COLORS['blue'])])
`

PR #146 (2023-03-09): added setup instructions for sudo-less RPi usage
Resolved the problem when using the SPI MOSI (GPIO 10) would result in unexpected behaviour when sending signals for a long periods. Turns out that Serial interface/ Serial port hardware has to be enabled otherwise the communication on SPI was being cut after few seconds. resolves #104 resolves #72 resolves #58
closed 2020-02-13 by MkLHX