adafruit/Adafruit_CircuitPython_NeoPixel #73: Odd issue, need some input

Mat-Moo, 2020-02-15:
Ok, I'm new with Python, but not with pixels or development. So I've been building some new fish tank lights, and written a tiny script to ramp up the lights in the morning and back down at night. This is running on a raspberry pi zero w with a 20a 5v psu connected to 5m of 5v ws2812b leds (I run several thousand on my house at Christmas, so had a few spare!).
The script is easy enough, and it's running once a minute via cron log:-
```
Fish-PI  2020-02-15 19:37:02.750930 set to [98, 98, 98]
Fish-PI  2020-02-15 19:38:02.805422 set to [94, 94, 94]
Fish-PI  2020-02-15 19:39:01.854232 set to [89, 89, 89]
Fish-PI  2020-02-15 19:40:01.911200 set to [85, 85, 85]
Fish-PI  2020-02-15 19:41:01.969742 set to [81, 81, 81]
Fish-PI  2020-02-15 19:42:02.048087 set to [76, 76, 76]
Fish-PI  2020-02-15 19:43:02.127141 set to [72, 72, 72]
Fish-PI  2020-02-15 19:44:02.182029 set to [68, 68, 68]
Fish-PI  2020-02-15 19:45:02.244382 set to [64, 64, 64]
Fish-PI  2020-02-15 19:46:02.316874 set to [60, 60, 60]
Fish-PI  2020-02-15 19:47:02.373193 set to [55, 55, 55]
Fish-PI  2020-02-15 19:48:02.464127 set to [51, 51, 51]
Fish-PI  2020-02-15 19:49:02.528094 set to [47, 47, 47]
Fish-PI  2020-02-15 19:50:02.584345 set to [42, 42, 42]
Fish-PI  2020-02-15 19:51:02.651717 set to [38, 38, 38]
Fish-PI  2020-02-15 19:52:02.703445 set to [34, 34, 34]
Fish-PI  2020-02-15 19:53:02.762262 set to [30, 30, 30]
Fish-PI  2020-02-15 19:54:02.825133 set to [26, 26, 26]
Fish-PI  2020-02-15 19:55:01.879388 set to [21, 21, 21]
Fish-PI  2020-02-15 19:56:01.956602 set to [17, 17, 17]
Fish-PI  2020-02-15 19:57:02.015411 set to [13, 13, 13]
Fish-PI  2020-02-15 19:58:02.078746 set to [8, 8, 8]
Fish-PI  2020-02-15 19:59:02.174539 set to [4, 4, 4]
Fish-PI  2020-02-15 20:00:02.294377 set to [0, 0, 0]
Fish-PI  2020-02-15 20:01:02.435078 set to [0, 0, 0]
```
So my calcs are correct, but when I send these via pixels.fill, I am getting some weird behaviour. Same when I use pixes[0] = (x,x,x).... 

BUT, only with some values, e.g. 62,62,62 is perfect, but then 64, 64, 64 is black?

What am I missing? Why am I getting this odd behavour?

ladyada, 2020-02-15:
hi mat - this sort of request is best handled by tech support - please contact the place you purchased the neopixels for technical assistance :)

Mat-Moo, 2020-02-15:
Why? If I connect these up and talk to them via an ESP module they work as expected. This is a software layer issue, so the question is which layer? 
If you write a simple app to ramp leds from all black to all white in 255 steps, does it work as expected?

Mat-Moo, 2020-02-16:
Actually one final thought, the data line is 3v, and the leds data line should be 5v, so wonder if I need a level shift in place and if that is causing a problem...

ladyada, 2020-02-16:
could be! :)

Mat-Moo, 2020-02-23:
So I added a 3.3v to 5v level shifter onto the Raspberry pi, and pleased to say I believe that has cured the problem. So closing and leaving for others reference.

closed 2020-02-23 by Mat-Moo