adafruit/Adafruit_CircuitPython_NeoPixel #85: Can't set brightness in NeoPixel()

LukeMoll, 2020-05-28:
```py
import board
import neopixel

# Causes error
pixels = neopixel.NeoPixel(board.A15, 8, brightness=0.5) 

# works fine
pixels = neopixel.NeoPixel(board.A15, 8, brightness=1.0) 
pixels.brightness = 0.5 
```
Setting `brightness` to than `1.0` in `NeoPixel()` causes the following error:
```
Traceback (most recent call last):
  File "main.py", line 5, in <module>
  File "neopixel.py", line 131, in __init__
  File "adafruit_pypixelbuf.py", line 103, in __init__
  File "adafruit_pypixelbuf.py", line 187, in brightness
  File "adafruit_pypixelbuf.py", line 206, in show
  File "neopixel.py", line 167, in _transmit
AttributeError: 'NeoPixel' object has no attribute 'pin'
```
I believe this is attributed to [self._transmit](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L166-L167) being called (referencing `self.pin`) by `pypixelbuf` as part of [super().__init__](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L131-L133), **before** `self.pin` is initialised on [line 135](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py#L135).

Tested with `Adafruit CircuitPython 5.3.0 on 2020-04-29; stm32f411ce-blackpill with STM32F411CE`, CircuitPython bundle 5.x 20200526.

tannewt, 2020-05-29:
@LukeMoll Would you mind creating a fix PR for this? I think you can either init the pin earlier or disable auto_write until everything is set.

LukeMoll, 2020-05-30:
Sure, I can have a go at that sometime next week :)

tannewt, 2020-06-01:
@LukeMoll Awesome! Thanks! Join our discord if you need any help (or just want to :-) ): https://adafru.it/discord

FoamyGuy, 2020-06-03:
This code is working properly for me on a Neo Trellis M4 with 5.4.0.beta and Neo pixel library out of today's library bundle:
```python
import board
import neopixel

pixels = neopixel.NeoPixel(board.NEOPIXEL, 32, brightness=0.05)
pixels[0] = (0,0,100)

while True:
    pass
```

I am not experiencing any crash or other issues from this code.

What device and version of Circuit Python are you using when you see this error?

LukeMoll, 2020-06-03:
@FoamyGuy I mentioned at the bottom of my issue:
> Tested with Adafruit CircuitPython 5.3.0 on 2020-04-29; stm32f411ce-blackpill with STM32F411CE, CircuitPython bundle 5.x 20200526.

I've got an ItsyBitsy M4 and Trinket M0 that I can try reproducing on later.

FoamyGuy, 2020-06-03:
Ah, whoops. I missed that on the first read. Thank you. The STM32 port is somewhat newer. I wonder if this is actually a symptom of an issue with that port rather than the neopixel library. 

I will test on a nrf52840 based device as well so we will cover all of the current main processor types between us.

Alasdair122, 2020-06-16:
Hi, I found same error when auto_write=True - could you verify this as well?

dherrada, 2020-06-30:
I actually ran into this testing someone's code from the forum, the brightness wouldn't change when auto_write was enabled

kevinjwalters, 2020-07-05:
Another issue from the forum which is likely to be this too: [Adafruit Forums: Help needed! REALLY basic!)](https://forums.adafruit.com/viewtopic.php?f=60&t=167031).

closed 2020-07-06 by tannewt