adafruit/Adafruit_CircuitPython_NeoPixel #86: Unable to control the last 2 unit of the led

kenken64, 2020-05-29:
[image]

This is my code 

`

import board
import neopixel
import os
import time

ORDER = neopixel.RGB

# GPIO pin 18 with 8 addressable leds
pixels = neopixel.NeoPixel(board.D18, 10, brightness=0.5, auto_write=False, pixel_order=ORDER)

# initialize all the env variables
BLINK_FEVER_FILE = os.getenv("BLINK_FEVER_FILE")
PROJ_DIR = os.getenv("PROJ_DIR")

# control the led strip off all the individual leds
def lightsOff():
    pixels.fill((0, 0, 0))

# control the led strip turn it to red all the individual leds
def lighUp():
    pixels.fill((255, 0, 0))

# control the led strip turn it to green all the individual leds
def noLighUp():
    pixels.fill((0, 255, 0))

while True:
    file = open(PROJ_DIR + BLINK_FEVER_FILE, "r")
    value = file.read()
    print(value)
    time.sleep(2)
    if value == '0':
        ightsOff()

    if value == '1':
        lighUp()

    if value == '2':
        noLighUp()
    pixels.show()

`

dhalbert, 2020-05-29:
Controlling only 3/4 of the NeoPixels would happen if you are trying to control RGBW NeoPixels as RGB. It sounds like your strip is RGBW (perhaps you order that by mistake). They need to have 4 color values sent for each pixel, but you are sending only 3.

Try setting `ORDER=neopixel.GRBW` or `RGBW`, or omit the `pixel_order` argument and set `bpp=4`.

In the latest code, `bpp` is only used to set the `order` if it is not otherwise specified.

kenken64, 2020-05-29:
I remove the order then everything is light up now :(
pixels = neopixel.NeoPixel(board.D18, 8, brightness=0.5, auto_write=False)

closed 2020-05-29 by kenken64