adafruit/Adafruit_CircuitPython_NeoPixel #92: Can't Run Neopixels On Pi Zero With Headphone Audio

compmodder26, 2020-10-09:
I'm trying to create a project with a pi zero that will use neopixels and also output audio via headphone jack circuit (https://learn.adafruit.com/adding-basic-audio-ouput-to-raspberry-pi-zero?view=all).  I'm running into an issue however where it appears that the neopixel library may be taking over PWM as when I run audio and the neopixel code together, I get weird screeching sound through the speaker.  I have the neopixel pin set to pin 12 and for the audio circuit I am using pins 18 and 13.  If I don't activate the neopixels from my code, the audio plays properly.  Is this going to be a limitation of the PI where I cannot generate 3 PWM signals?  Are you aware of a way to make the neopixel library utilize software PWM?

compmodder26, 2020-10-09:
Tried changing to a mono setup by updating the dtoverlay setting from:
dtoverlay=pwm-2chan,pin=18,func=2,pin2=13,func2=4
to:
dtoverlay=pwm,pin=13,func=4

Verified that audio still worked properly with that setup.
However, if I try to run the neopixel code and the audio together still, I get garbage sound.  Thought this solution might work okay because 18 and 12 share PWM0 while 13 and 19 share PWM1, but that doesn't seem to have made a difference.

ladyada, 2020-10-09:
correct, they both use the PWM DMA, so you can't use both! you could try SPI pins

compmodder26, 2020-10-09:
I hadn't seen the SPI option for the circuitpython neopixel software.  That's awesome.  So to clarify from this tutorial (https://learn.adafruit.com/circuitpython-neopixels-using-spi/overview), it should work the same on the pi correct?  And I only need to use MOSI (pin 10)?

ladyada, 2020-10-09:
possibly - worth a shot!

compmodder26, 2020-10-09:
Tried it, and it worked a treat!  Thanks for all your help!

closed 2020-10-09 by ladyada