adafruit/Adafruit_CircuitPython_NeoPixel #93: NeoPixel length within Python library

AndrewMohawk, 2020-10-12:
It looks like you cannot set the neopixel strip length within this Python library, but it "updateLength" function seems to be available elsewhere: https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L126

ladyada, 2020-10-13:
correct, you need to set the length on initialization! you could always set it longer and not use all the pixels

closed 2020-10-13 by ladyada

AndrewMohawk, 2020-10-13:
I'm building something that allows people to configure their system dynamically by watching the changes and they might have strips that are different sizes, so I was hoping the updateLength function would be exposed as well since its in the other libraries. I can just re-initialize on changes instead.