adafruit/Adafruit_CircuitPython_NeoPixel #94: reinitialising NeoPixel library after calling deinit() does not work

AndrewMohawk, 2020-10-13:
Heya,

I need to re-initialise the NeoPixel library and it looks like I should be calling deinit() after I am done, but once I have done this, when I re-initialise the library the strip is no longer responsive:

`>>> import board, neopixel
>>> x = neopixel.NeoPixel(board.D18,300,auto_write=False); x.fill((255,0,0)); x.show();
>>> #Strip showing red, everything is good
>>> x.deinit()
>>> #Strip is off, looks okay
>>> x = neopixel.NeoPixel(board.D18,300,auto_write=False); x.fill((0,255,0)); x.show();
>>> #No errors but strip is not responsive
`

I suspect there could be an issue in how it resets the pin, functionally the object works without any errors, but it no longer lights up the LEDs.

Thanks for all the work on the lib, its super useful!

AndrewMohawk, 2020-10-13:
Some additional information, this is running on a pi4 with these versions of the libraries:
```$ sudo pip3 list | grep -i adafruit
Adafruit-Blinka                   5.0.1      
adafruit-circuitpython-neopixel   6.0.0

AndrewMohawk, 2020-10-14:
I've upgraded all the libraries and all the packages on my pi4 as well and am now running the following with the same issues: 
```
Adafruit-Blinka                   5.5.2      
adafruit-circuitpython-neopixel   6.0.0      
adafruit-circuitpython-pypixelbuf 2.2.0

FoamyGuy, 2020-10-14:
I tried this out on a `Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Circuit Playground Bluefruit with nRF52840` with this slightly modified version of your reproducer: 
```python
import board, neopixel, time
x = neopixel.NeoPixel(board.NEOPIXEL,10,auto_write=False, brightness=0.1)
x.fill((255,0,0))
x.show();
time.sleep(3)
#Strip showing red, everything is good
x.deinit()
#Strip is off, looks okay
time.sleep(2)
x = neopixel.NeoPixel(board.NEOPIXEL,10,auto_write=False, brightness=0.1)
x.fill((0,255,0))
x.show()

while True:
    pass
```

In my case the deinit appears to be working properly. The Neopixels are correctly showing red, then off, then green.

The main changes I made to the script were splitting a few of the lines into multiples for clarity, changing the pin and pixel count to match my device, and adding some `time.sleep()` calls so it can run outside of the REPL and still produce meaningful results.  To the best of my knowledge none of that should have had an effect on this issue. 

My guess is that this might be limited to Raspberry Pi / Blinka based devices.

AndrewMohawk, 2020-10-14:
Heya FoamyGuy,

Thanks a lot for testing this on your side as well, I tried the above code but after the deinit I get no response from the strip, even using just a strip of 10 with the low brightness :( 

-AM