adafruit/Adafruit_CircuitPython_NeoPixel #96: deinit() on raspberry pi does not allow successful reinitialzation

jdimpson, 2020-10-25:
The following code does not produce desired result on a Raspberry Pi Zero W running Raspbian 9.13:
```
pi@nullpi:~/bin $ sudo python3
Python 3.5.3 (default, Jul  9 2020, 13:00:10) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import neopixel
>>> np = neopixel.NeoPixel(board.D18, 16)
>>> np.n
16
>>> for i in range(np.n): np[i] = (255,0,0)      # works as expected, 16 red lit neopixels
... 
>>> np.deinit()                                              # neopixels go dark
>>> np=None
>>> np = neopixel.NeoPixel(board.D18, 8)
>>> np.n
8
>>> for i in range(np.n): np[i] = (255,0,0)    # no effect, neopixels stay dark
... 
```
When run, the above code lights up the first 16 neopixels as red, which then go dark at the deinit() call. But assignment to the second neopixel object does not turn on any neopixels. It doesn't seem to matter how many neopixels are specified in either the first or second object instantiations. There are no errors thrown nor anything seemingly relevant in /var/log/messages or dmesg. 

Similar code running on a Gemma M0 with a 5.X build of CircuitPython works as expected.

jdimpson, 2020-10-25:
I forgot to mention that this is after I ran `sudo pip3 install rpi_ws281x adafruit-circuitpython-neopixel`to ensure it was up to date as of the time of the issue being opened.

ladyada, 2020-10-26:
hi is this is a duplicate of
https://github.com/adafruit/Adafruit_Blinka/issues/355
? closing this one so you can sub to the other thread!

closed 2020-10-26 by ladyada

jdimpson, 2020-10-26:
Looks like it. Thank you!