adafruit/Adafruit_CircuitPython_NeoPixel #98: neopixel_cleanup should be called on deinit()

jamesrusso, 2020-12-12:
The neopixel_cleanup from the underlying neopixel_write module method should be called on deinit(). Without this call it is not possible to change the pixel lengths in the same process. The neopixel_cleanup method is called atexit, but it is likely desirable to have it just happen on the deinit method. 

Ex: 

The LEDs will never turn WHITE in this example: 

```
import neopixel # pylint: disable=import-error
import time
import board

gpio_port = board.D18
pixels = neopixel.NeoPixel(gpio_port, 5, auto_write=False)
## Turn the first 5 pixels RED.
pixels.fill((255,0,0))
pixels.show()
time.sleep(3)
## Turn the first 3 pixels WHITE.
pixels.deinit()
pixels = neopixel.NeoPixel(gpio_port, 3, auto_write=False)
pixels.fill((255,255,255))
pixels.show()
time.sleep(3)
```

This will likely require an upstream change to circuit python, as the neopixel_cleanup method is not exposed in the non-board specific python file.

jamesrusso, 2020-12-12:
I've opened this: https://github.com/adafruit/Adafruit_Blinka/issues/378

jamesrusso, 2020-12-12:
I've created PR #99 to resolve this along with https://github.com/adafruit/Adafruit_Blinka/pull/379 on the NeoPixel module.

tekktrik, 2024-03-28:
@tannewt it sounds like the underlying issue here was solved in another way, but just wanted to check.

tannewt, 2024-03-29:
@tekktrik I have no context. Are you suggesting I close this?

jamesrusso, 2024-03-29:
As the original opener, I think this was closed as it was discussed in upstream issue and work around was determined

tekktrik, 2024-03-29:
@tannewt yes, sorry.  I was looking at open issues and saw this and thought it might be considered closed, but wasn't entirely sure.

tannewt, 2024-04-01:
Ok thank you both!

closed 2024-04-01 by tannewt