adafruit/Adafruit_CircuitPython_NeoPixel #101: Breaking change somewhere in recent Raspberry Pi OS update

bluesign2k, 2021-01-22:
Yesterday, did an upgrade on some Raspberry Pi 3B+ that use this library to drive a newpixel ring as part of an interface application. After the upgrade, the rings stopped functioning - even the example scripts provided in the repo fail to display properly. On the off-chance that something went wrong with the upgrade, I tried a fresh image on a fresh SD card and installed only this library but got the same results.
Of the 24 neopixels in the ring. only the first 4 or 5 do anything and they tend to be flickering random colours, mainly greens, blue and sometimes white.
It's not a hardware fault as if I transfer the ring to another device that hasn't been upgraded it works perfectly.

ladyada, 2021-01-22:
we haven't had any changes to this repo in quite a while (other than documentation)
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/commits/master
maybe check your voltages, try powering the ring from 3.3V
@makermelissa it could also be due to the SPI DMA firmware update/bug?

makermelissa, 2021-01-22:
It's possible, though I don't think this library uses SPI. @bluesign2k, when you say "did an upgrade", do you mean you updated the Raspberry Pi Linux environment or Adafruit-Blinka or both?

bluesign2k, 2021-01-22:
Sorry, yes, when I said I 'did an upgrade', I meant I did a 'sudo apt upgrade', not updating the library - I wasn't referring their being a breaking change in this library, but more that this library now seems to be incompatible with at least the latest official Raspberry Pi OS release. 
@makermelissa Initially I updated the RPi Linux environment, but I effectively did both in a later test when I tried with a fresh Linux image and a fresh install of this library....and it still didn't work.
I'll try powering the ring with 3.3v, but I'm not sure it'll sort it.

makermelissa, 2021-01-22:
Thanks, it helps to get an idea of where to check for things to fix/recreate. You can try this page and see if it helps: https://learn.adafruit.com/adafruit-mini-pitft-135x240-color-tft-add-on-for-raspberry-pi/kernel-module-troubleshooting

bluesign2k, 2021-01-28:
I've setup a new Pi3A+ with the latest OS image and as I'd experienced previously, this failed to work. This was kernel version `1.20210108-1`. `uname -a` gave:
`Linux pi3a 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux`

I then followed the instruction in the guide that @makermelissa posted to roll back the kernel version to `20201126-1`, and that made no difference - the ring still just glows random green/white colours randomly. `uname -a` gives:
`Linux pi3a 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux`

I also tried on an old Pi3B+ that I'd been using for something else - I upgraded everything including the kernel up to `1.20210108-1` and this works fine. I'm stumped as to why this one is working, but thre's a clearly a difference.

makermelissa, 2021-01-28:
Ok thanks, that helps. The other issue seemed to only affect the Pi 4 as well.

bluesign2k, 2021-01-30:
Its taken a while, but I've found the package that causing the problem, albeit not specifically why (and tbh, I'm not going to try to look into why either).

I started by downloading an old OS image back from the middle of last year (`2020-08-20-raspios-buster-armhf`) and slowly went through the list of available upgrades until it died. I eventually found that if I put a hold on all the raspberry pi core packages (sudo apt-mark hold libraspberrypi* raspberrypi*) but then removed the hold only on `raspberrypi-ui-mods` then `pulseaudio` was one of the packages it installed, and it was this that killed it.

I've since tried removing this package both on a system that I broke last week when I did a full upgrade, and on a fresh image of the latest OS download, and both now work.  So for now, if anyone else experience the problem, just `sudo apt remove pulseaudio` and reboot :)

o-bri, 2021-03-15:
@bluesign2k 
Thanks for sharing your solution.
I also had the problem, that with the actual Raspi OS the pixel behaved crazy.

removing pulseaudio did the trick for me too.

smart-fun, 2021-03-31:
Hi, I also have a strange behavior here, on a Raspberry 3B+.

I can change the colors of the 24 leds of the ring 1 time, but then they are (almost) always white. The strange part is that if I move a window quickly on the rasp (like the terminal or Thonny IDE) then the leds are updated with the correct colors. I have tried to remove pulseaudio and reboot but no change.

@makermelissa I think the library uses SPI as if I disable it on the pi I got a runtime error.

Here is a simple code which is supposed to switch between a blue and a red ring every 2 seconds (but does blue then white):

```python
# see https://cdn-learn.adafruit.com/downloads/pdf/neopixels-on-raspberry-pi.pdf

import board
import neopixel
from time import sleep

pixels = neopixel.NeoPixel(board.D10, 24, brightness=0.3, auto_write=False, pixel_order=neopixel.GRB);

while True:
    
    pixels.fill((0,0,255))
    pixels.show()
    sleep(2)

    pixels.fill((255,0,0))
    pixels.show()
    sleep(2)
```

mikeesto, 2021-09-27:
> I've since tried removing this package both on a system that I broke last week when I did a full upgrade, and on a fresh image of the latest OS download, and both now work. So for now, if anyone else experience the problem, just `sudo apt remove pulseaudio` and reboot :)

Thanks, I ran into exactly the same issue today. Removing Pulse Audio worked for me as well but I actually want it. I ended up using the [NeoPixel SPI library](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel_SPI) as a workaround - it plays nicely with Pulse Audio.

cohnt, 2021-12-16:
Removing pulseaudio didn't work for me, but completely disabling the audio on the raspberry pi did.

According to the instructions on this site: https://forums.raspberrypi.com/viewtopic.php?t=260568

1. `sudo nano /etc/modprobe.d/raspi-blacklist.conf`
2. Add tihis line to the file, and save: `blacklist snd_bcm2835`
3. Reboot