adafruit/Adafruit_CircuitPython_NeoPixel #107: Support HSV colors

sermelo, 2021-03-13:
For those who are not aware of what HSV is: https://en.wikipedia.org/wiki/HSL_and_HSV

I think it will be really useful to work with HSV colors, to operate the leds in a different way. I have already done some changes in a fork, and I have created a draft Pull Request:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/pull/106

The pull request is just a proposal to get some feedback from you and know if this is something you would be interested in. If it is OK to you, I can work in improving it, add documentation, support HSVW, etc.

ladyada, 2021-03-13:
hihi we support HSV thru fancyled library - check it out!

sermelo, 2021-03-15:
That looks good, I will close the PR.

closed 2021-03-15 by sermelo