adafruit/Adafruit_CircuitPython_NeoPixel #110: Setup Errors - Can't install on Raspberry Pi 4

JonLevin25, 2021-04-16:
I'm trying to install from `pip3` on a Raspberry Pi 4 running RetroPi / Raspbian Buster.

Setup collects packages, and then fails on `setup.py` stating:
```
    ******************************************************************************
    * Setup can't determine the value of PAGE_SIZE on your system, so it will
    * default to 4096 which may not be correct.
    *
    * Please report this message and your operating system info to the package
    * maintainer listed in the README file.
    ******************************************************************************
```

More diagnostics/troubleshooting tried:
* I've managed to install the library previously on a Raspberry Pi 3 (Which I think was also running Buster)
* I've managed to install other python packages on the Pi 4 using `pip3 install` i.e. `tornado`, `colorzero` etc.
* I've run `sudo apt update && sudo apt upgrade` + rebooting- same issue
* I've tried python 2 (`pip install pip install adafruit-circuitpython-neopixel`) - same error

I'm not sure if this is a system configuration issue on my end or some hardware/OS incompatibility, but would really like to get Neopixels up and running on the Pi - any help would be appreciated

Thanks

ladyada, 2021-04-16:
paste the full output and command you are running with verbose output, because that error isnt coming from this library

JonLevin25, 2021-04-17:
Attaching Verbose output - The command was `pip3 install --verbose adafruit-circuitpython-neopixel`.
I also installed `wheel` in between to see if it would help, no dice so far.

```
Using pip 21.0.1 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)
Defaulting to user installation because normal site-packages is not writeable
Created temporary directory: /tmp/pip-ephem-wheel-cache-_civ08p9
Created temporary directory: /tmp/pip-req-tracker-wdl6ffh0
Initialized build tracking at /tmp/pip-req-tracker-wdl6ffh0
Created build tracker: /tmp/pip-req-tracker-wdl6ffh0
Entered build tracker: /tmp/pip-req-tracker-wdl6ffh0
Created temporary directory: /tmp/pip-install-dmof3wpr
1 location(s) to search for versions of adafruit-circuitpython-neopixel:
* https://pypi.org/simple/adafruit-circuitpython-neopixel/
Fetching project page and analyzing links: https://pypi.org/simple/adafruit-circuitpython-neopixel/
Getting page https://pypi.org/simple/adafruit-circuitpython-neopixel/
Found index url https://pypi.org/simple
Looking up "https://pypi.org/simple/adafruit-circuitpython-neopixel/" in the cache
Request header has "max_age" as 0, cache bypassed
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/adafruit-circuitpython-neopixel/ HTTP/1.1" 304 0
  Found link https://files.pythonhosted.org/packages/13/be/247b778e7777c548b31b33761879cd1c2d2350402c2209abf7e883cba3fd/adafruit-circuitpython-neopixel-3.3.0.tar.gz#sha256=30fde13c9c47dc4abdbc2a5c0a9f0dc318016a446cd84e4a7ea0482a2ff7105d (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.0
  Found link https://files.pythonhosted.org/packages/cf/8d/240a15027b80c1204f903d51a5c72046dc326f3151d147bb1b0fc230e357/adafruit-circuitpython-neopixel-3.3.1.tar.gz#sha256=bf49e8a18d1065f02189a92a859bf576213694229e7a75bfeef2ebfcdedaedff (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.1
  Found link https://files.pythonhosted.org/packages/2a/51/26062159de1de45130860c30092b2515d5e17c3345b1c0289c2d2a4d1485/adafruit-circuitpython-neopixel-3.3.2.tar.gz#sha256=23f234860a5cbabb42348d00c346af9bdc8b00787ed7a6e97fb732fadbf93886 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.2
  Found link https://files.pythonhosted.org/packages/2a/89/ac49d02b5ff22db2a8db0bbdb4a66482c5fa1b38df5d73b1e455b5270a06/adafruit-circuitpython-neopixel-3.3.4.tar.gz#sha256=30a02c309e8ef851395e6aa75987062d8919e10a9a51c25ac3b266f6f64ddb61 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.4
  Found link https://files.pythonhosted.org/packages/d0/ab/d60f104e52bcfcfa29762aad8767d964cc5809868fed9969def0139b5728/adafruit-circuitpython-neopixel-3.3.5.tar.gz#sha256=a0fe74ad1f506cfb8f14c746901dd1c29e963d45e417687301c3d84a7e24b8ea (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.5
  Found link https://files.pythonhosted.org/packages/7c/cb/b4ecc18843d80689e8e7bb242ab065b4ec05a1f33eb9344fe7f39600e867/adafruit-circuitpython-neopixel-3.3.6.tar.gz#sha256=2bb83a94ed7701db92738e769d205f70cda6acbdd08db1a19f5ad14a5ffe1df2 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.6
  Found link https://files.pythonhosted.org/packages/f8/f3/a9c114f66ae5e5c4da390991109442c33baa76a69e615bfdca5bc8c668bf/adafruit-circuitpython-neopixel-3.3.7.tar.gz#sha256=b280f3ef9ecf4f307cd7ebf1952ee802ee050faf96ff3c70d61e95ed609dc609 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.3.7
  Found link https://files.pythonhosted.org/packages/7c/23/7bdba7f48f2129dea0bb384b3ebaad0a579114595c9e3e5e02c51d75138a/adafruit-circuitpython-neopixel-3.4.0.tar.gz#sha256=a961eb84041bc21f33750b4eb4604edf5b7d00fbe7687969a1699d5d9e75534b (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 3.4.0
  Found link https://files.pythonhosted.org/packages/af/54/249a17ee20807574af352e9e85a43a46303f8915314b3ccd8f246a401ffe/adafruit-circuitpython-neopixel-4.0.0.tar.gz#sha256=8f30fe697524d0b9bffdf3d87970744d1a7121b8d1c4fae0ecca5fada25eaf3f (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 4.0.0
  Found link https://files.pythonhosted.org/packages/21/d8/d65c83ac85d7b86686b004f774e507ee5185f09af6540ec1b3bb3a702b30/adafruit-circuitpython-neopixel-4.1.0.tar.gz#sha256=1aaebcc1eaf157b0811d5d5568b2843741fcb6ad78d165d3a7bbfefe2882276c (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 4.1.0
  Found link https://files.pythonhosted.org/packages/3e/0f/9cae58f5c2483ffd9819917b3dd1cee1de5669361a66afcef001c3bfbd74/adafruit-circuitpython-neopixel-5.0.0.tar.gz#sha256=668b8e79d46a2deb828c4b05727d0723a36af4295478f21e66f06de543718219 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 5.0.0
  Found link https://files.pythonhosted.org/packages/7e/d7/a39738c05a786febf309f9f147b015cc5381096cad6d5b4ee833a1a52de3/adafruit-circuitpython-neopixel-5.1.0.tar.gz#sha256=055ae199d6c28cb7b51c5fb56cddbee0c51ca672828161065e691de4316e5328 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 5.1.0
  Found link https://files.pythonhosted.org/packages/3a/55/500719e7da4c892b12a55affba47e2cee142302ba1f3d30ca264c72c2387/adafruit-circuitpython-neopixel-5.1.1.tar.gz#sha256=42ac6ae82ccb1c51c051d48c85ed4e3074b260fb572cb59c3e0bf3712bc71a59 (from https://pypi.org/simple/adafruit-circuitpython-neopixel/), version: 5.1.1
[... cut ...]
  copying build/lib/adafruit_blinka/board/raspberrypi/raspi_40pin.py -> build/bdist.linux-armv7l/wheel/adafruit_blinka/board/raspberrypi
  creating build/bdist.linux-armv7l/wheel/adafruit_blinka/board/lubancat
  copying build/lib/adafruit_blinka/board/lubancat/__init__.py -> build/bdist.linux-armv7l/wheel/adafruit_blinka/board/lubancat
  copying build/lib/adafruit_blinka/board/lubancat/lubancat_imx6ull.py -> build/bdist.linux-armv7l/wheel/adafruit_blinka/board/lubancat
  copying build/lib/adafruit_blinka/board/udoo_x86ultra.py -> build/bdist.linux-armv7l/wheel/adafruit_blinka/board

JonLevin25, 2021-04-17:
Someone on StackOverflow suggested installing `python3-dev`. Not entirely sure why it was necessary, but it worked :)

https://stackoverflow.com/questions/67133353/cant-install-pip-package-on-raspberry-pi-4-adafruit-circuitpython-neopixel

closed 2021-04-19 by tannewt