adafruit/Adafruit_CircuitPython_NeoPixel #122: import board error with Raspberri Pi Zero 2 WH

echo-alfa, 2022-01-12:
I am getting the following error when attempting to import board using a Raspberri Pi Zero 2 WH. 

File "testpixel.py", line 6, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 210, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None

This very same program runs absolutely fine on a Pi Zero WH.

I have made sure all updates are done but I suspect that the Pi Zero 2 has not yet been added to board.py?

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 00000000c2fb4c94
Model           : Raspberry Pi Zero 2 Rev 1.0

echo-alfa, 2022-01-12:
I'm using

sudo pip3 install rpi_ws281x adafruit-circuitpython-neopixel

Which worked with the original Zero W?

caternuson, 2022-01-12:
That's for installing NeoPixel support. How did you install Blinka?
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

echo-alfa, 2022-01-12:
It was my understanding that that also installed blinka?

 $ sudo pip3 install rpi_ws281x adafruit-circuitpython-neopixel
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: rpi_ws281x in /usr/local/lib/python3.7/dist-packages (4.2.5)
Requirement already satisfied: adafruit-circuitpython-neopixel in /usr/local/lib/python3.7/dist-packages (6.0.2)
Requirement already satisfied: Adafruit-Blinka in /usr/local/lib/python3.7/dist-packages (from adafruit-circuitpython-neopixel) (6.2.2)
Requirement already satisfied: adafruit-circuitpython-pypixelbuf>=2.0.0 in /usr/local/lib/python3.7/dist-packages (from adafruit-circuitpython-neopixel) (2.2.3)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (0.52.9)
Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (3.1.1)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (1.1.8)
Requirement already satisfied: sysv-ipc>=1.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (1.1.0)
Requirement already satisfied: RPi.GPIO in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (0.7.0)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-neopixel) (1.1.1)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-neopixel) (3.5)

echo-alfa, 2022-01-12:
$ sudo pip3 install Adafruit-Blinka
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: Adafruit-Blinka in /usr/local/lib/python3.7/dist-packages (6.2.2)
Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (3.1.1)
Requirement already satisfied: sysv-ipc>=1.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.1.0)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.1.8)
Requirement already satisfied: RPi.GPIO in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.7.0)
Requirement already satisfied: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.52.9)
Requirement already satisfied: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (4.2.5)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (3.5)
Requirement already satisfied: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (1.1.1)

caternuson, 2022-01-12:
It can, but it looks like some other previous install was already done. It's finding and using an older version of Platform Detect that does not have the Pi Zero W 2 rev codes:
```
Requirement already satisfied: Adafruit-PlatformDetect>=3.1.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-neopixel) (3.1.1)
```
Try this:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/faq-troubleshooting#update-blinka-slash-platform-libraries-3030915-1

closed 2022-01-12 by echo-alfa