adafruit/Adafruit_CircuitPython_NeoPixel #129: Neopixels stop responding after some time

jonesthefox, 2022-06-05:
Hello there! ^^

I use my neopixel matrix 64 rgbw as a light source for my plant growing project. 

I have a light script, that implements sunrise, noon and sunset. 
basically, the light dims up, sleeps while noon (16 hours in my case) and then it dims down again. 

The `pixels = neopixel.NeoPixel(...)` was defined on top of the script, after the includes and some config. It was working fine, dimming the light up and down. But after updating to v6.3.0 (i guess, but nothing fancy was changed there) the neopixels won't respond after 16h sleep of the script.

Because I didn't want to upset my basil, i was not able to test it thoroughly, but i guess that 15-20 minutes after the last change to the neopixels is sent, the library silently fails to change the neopixels any further (the log says it dims down, but nothing happens at all).

I circumvent this, by invoking `neopixel.NeoPixel(...)` when i need it, as you can see here in my commit to my light daemon [here](https://github.com/jonesthefox/growbox/commit/b3df84ea120361eca43a46cb0fe81143e6b3e47d) 

I believe that this is a bug in the library, but I am also willing to admit that my code is bad :-)

dhalbert, 2022-06-05:
Thanks -- some questions:

What board are you using this on?
What version of CircuitPython are you using?
Are all your libraries up to date?

jonesthefox, 2022-06-05:
The Board is an Raspberry Pi 3 Model B Rev 1.2 and i run the script with python 3.9.2. The libraries are all up to date. Unfortunately i can't tell what version adafruit-circuitpython-neopixel was, before i updated it (blindly) to 6.3.0..