adafruit/Adafruit_CircuitPython_NeoPixel #132: Number of led in the strip really affect the speed.

gumush, 2022-08-04:
I use simple led move but it's very very slow in 1800leds ( just 30m 60led/meter ) 

def run(position ,color , speed , size  ):
    turnoff()
    for i in range(position):
        wait = 1/(speed*60)
        clean = int(60*size)
        print(f'Clean:{clean} Wait:{wait} I:{i}')
        pixels[i]=[255,0,0]
        if i-clean>0:
            pixels[i-clean]=[0,0,0]
        else:
            pixels[0]=[0,0,0]
        # time.sleep(0.001)
    turnoff()

Is there any way to speed this thing up ? 

I guess in microcontrollers this would be better ?

tremby, 2023-01-03:
I came here hoping to find out about performance and any limits. Your post is concerning. The project I'm planning would have around 830 LEDs, and I was hoping to control it with a Pi Nano. If anyone could chime in with whether this is realistic or not it would be appreciated!

pathnirvana, 2023-02-07:
since python is an interpreted language it is slow when updating large number of leds with a high refresh rate. It is best to stick with C++ libraries in arduino

dhalbert, 2023-02-07:
You can control when a write to the strip happens. Set `auto_write` to false in the constructor, and call `show()` when you want to send the state to the strip. I didn't do that below because I don't know how you want it to look.

Also, in the loop above, factor out the constant values. Create them once outside the loop. For instance:
```py
RED = (255, 0, 0)
OFF = (0, 0, 0)
def run(position ,color , speed , size  ):
    turnoff()
    wait = 1/(speed*60)  # only needs to be computed once
    clean = int(60*size)  # only needs to be computed once
    for i in range(position):
        #print(f'Clean:{clean} Wait:{wait} I:{i}') # Don't print if you don't need to.
        pixels[i]=RED
        if i-clean>0:
            pixels[i-clean]=OFF
        else:
            pixels[0]=OFF
        # time.sleep(0.001)
    turnoff()
```

FoamyGuy, 2024-12-23:
I tested this on a raspberry pi pico to get some more specific numbers around it. I used variations of this script to test:
```
num_pixels = 1800
ORDER = neopixel.GRB
pixels = neopixel.NeoPixel(
    pixel_pin, num_pixels, brightness=0.2, auto_write=True, pixel_order=ORDER
)

before = time.monotonic()
print(before)
for i in range(num_pixels):
    pixels[i] = 0xff0000
    # if i % 10 == 0:
    #     pixels.show()
#pixels.show()
print(f"took {time.monotonic() - before} seconds")
```

With `auto_write=True` it took 100.196 seconds to complete the loop updating the neopixels one at a time.

With `auto_write=False` and only a single call to `show()` at the end of the loop it took 0.149902 seconds only. 

With `auto_write=False` and a call to `show()` every 10 iterations it took 10.02 seconds.

Closing this issue. The number of calls to `show()` and by relation the `auto_write` configuration have a much larger effect on the speed than the number of pixels in the strip. If you update once on every then longer strands will take more time, but if disable `auto_write` and only call `show()` as needed it can still go fast on with large numbers of LEDs.

closed 2024-12-23 by FoamyGuy