adafruit/Adafruit_CircuitPython_NeoPixel #134: No Module Named adafruit_pixelbuf

polyphonic13, 2022-08-14:
The adafruit_pixelbuf module is not found when running on a 2022 Raspberry Pi 4, with Raspbian OS. 

Steps to Reproduce: 

1. Install with `sudo pip3 install rpi_ws281x adafruit-circuit-python-neopixel`
2. Create python script (this was named "eye_lights01.py) that leverages NeoPixel with `import neopixel` (or `from neopixel import *` -- tried multiple demo scripts)
3. Run script with `sudo python3 eye_lights01.py`

*Expected Result:* AdaFruit lights do lighting things
*Actual Result:* bug thrown from "neopixel.py" script:

```Traceback (most recent call last):
  ...
  File "usr/local/lib/python3.9/dist-packages/neopixel.py", line 20 in <module>
    import adafruit_pixelbuf
ModuleNotFoundError: No module named 'adafruit_pixelbuf'
```

Neradoc, 2022-08-14:
There's a typo (there is no `-` in circuitpython) but I assume that's only here, since I can reproduce the issue.
Tested in a venv, it also did not install Blinka (which you probably already had).
```
python3 -m venv venv
source venv/bin/activate
pip install rpi_ws281x adafruit-circuitpython-neopixel
```
```py
>>> import neopixel
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/circuit/tests/venv/lib/python3.7/site-packages/neopixel.py", line 16, in <module>
    import board
ModuleNotFoundError: No module named 'board'
>>> 
```
You can install the missing dependency manually:
```
sudo pip3 install adafruit-circuitpython-pixelbuf
```

polyphonic13, 2022-08-14:
Thank you! `adafruit-circuitpython-pixelbuf` was the dependency name that I couldn't figured out. I'd tried `adafruit-pixelbuf` and `pixelbuf` with no luck. 

> There's a typo (there is no - in circuitpython)
Correct. That was only in this post. Edited original to correct.

closed 2022-08-17 by tekktrik