adafruit/Adafruit_CircuitPython_NeoPixel #144: Example error consult

zkbutt, 2022-11-21:
[image]

Traceback (most recent call last):
  File "....t001.py", line 4, in <module>
    pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)
AttributeError: module 'board' has no attribute 'NEOPIXEL'

zkbutt, 2022-11-21:
[image]
How do we solve this? Can we disable this restriction

Kasperror, 2022-12-15:
> [image] How do we solve this? Can we disable this restriction

If the board you're using is Raspberry Pi the solution sits in my PR #146 which authors did not approved yet.

Kasperror, 2022-12-15:
> [image]
> 
> Traceback (most recent call last): File "....t001.py", line 4, in pixels = neopixel.NeoPixel(board.NEOPIXEL, 1) AttributeError: module 'board' has no attribute 'NEOPIXEL'

use hardware port instead, just like some examples show - `board.D18` or `board.D10` depending on the hadware you're using

caternuson, 2022-12-15:
You can only use `board.NEOPIXEL` for boards that have a dedicated pin tied to a NeoPixel. General way to see what pins exist:
```python
import board
dir(board)
```

A Raspberry Pi does *not* have a NEOPIXEL pin.

For the permissions issue, see PR linked above.

Closing since not an actual library issue.

closed 2022-12-15 by caternuson