adafruit/Adafruit_CircuitPython_NeoPixel #149: ROS2: memory leak

keukenrol, 2023-04-22:
Hi

when implementing the neopixel library in a ROS2 node, I get the following error:

`Failed to create mailbox device
: Operation not permitted
Traceback (most recent call last):
  File "/home/ubuntu/turtlebot3_ws/src/install/test_nodes/lib/test_nodes/ledSub", line 11, in <module>
    load_entry_point('test-nodes==0.0.0', 'console_scripts', 'ledSub')()
  File "/home/ubuntu/turtlebot3_ws/src/install/test_nodes/lib/python3.8/site-packages/test_nodes/ledSub.py", line 82, in main
    pixel_node = pixelNode()
  File "/home/ubuntu/turtlebot3_ws/src/install/test_nodes/lib/python3.8/site-packages/test_nodes/ledSub.py", line 44, in __init__
    self.pixels.show()
  File "/usr/local/lib/python3.8/dist-packages/adafruit_pixelbuf.py", line 205, in show
    return self._transmit(self._post_brightness_buffer)
  File "/usr/local/lib/python3.8/dist-packages/neopixel.py", line 180, in _transmit
    neopixel_write(self.pin, buffer)
  File "/usr/local/lib/python3.8/dist-packages/neopixel_write.py", line 38, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 82, in neopixel_write
    raise RuntimeError(
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
`

This is triggered when calling the function self.pixels.show(), which is inside a timer callback of a node.
Any idea how to solve?

Thanks!

ealaimo18, 2024-04-11:
Did you ever solve this? Running into the same issue now running that library on Ros2.

keukenrol, 2024-04-22:
> Did you ever solve this? Running into the same issue now running that library on Ros2.

No, we ended up running the script separately.