adafruit/Adafruit_CircuitPython_NeoPixel #151: not support more than 30 LEDs bug

xgqfrms, 2023-05-07:
#  not support more than 30 LEDs bug

## environment

Raspberry Pi `3B`

Python `3.9.2`

one WS2812B RGB LED strip  of `1M` long with `60 LEDs`

## packages

```sh
$ sudo pip3 install rpi_ws281x

$ sudo pip3 install adafruit-circuitpython-neopixel

$ sudo pip3 install --force-reinstall adafruit-blinka
# equal to
$ sudo python3 -m pip install --force-reinstall adafruit-blinka
```

## codes

```py
#!/usr/bin/env python3
# coding: utf8

# neo pixel ✅
import board
from time import sleep
import neopixel

# 60 个LEDs
pixels = neopixel.NeoPixel(board.D18, 60)
while True:
  for x in range(0, 60):
    pixels[x] = (255, 0, 0)
    sleep(0.1)

```

FoamyGuy, 2023-05-08:
I don't have an answer for certain, but a few questions and observations that may be related:

- What is your power source? and what brightness level are the LEDs on. Turning on many pixels brightly can use a lot of power, if your source cannot handle the amount needed it could result in some in the strand not turning on, or other issues.
- Is the photo in your post taken while running the exact code that is also in your post? If so, it's worth noting that the code seems to be attempting to set the pixels all to (255,0,0) which would be red (depending on color order). But in the photo the lights are lit up many different colors not all the same color, red or otherwise. If the pixels are lighting up different colors from what the code is setting to them it makes me wonder if bytes per pixel configuration is not matching your strip, maybe your strip contains 4 bytes with RGBW?

xgqfrms, 2023-05-09:
@FoamyGuy Thanks for your time!

1. just use the Raspberry Pi 3B `5V` GPIO PIN
2. I didn't configure the brightness level for it. (the brightness level should be the default value)

I will try your suggestions later.

xgqfrms, 2023-06-01:
# update

> 60 LEDs with 255 brightness level need to add an external Power Supply.

```py
#!/usr/bin/env python3
# coding: utf8

from rpi_ws281x import *

from time import sleep

# LED strip configuration:

# Number of LED pixels.
LED_COUNT = 60

# GPIO pin connected to the pixels (18 uses `PWM`).
LED_PIN  = 18
# GPIO pin connected to the pixels (10 uses `SPI` /dev/spidev0.0).
# LED_PIN = 10

# LED signal frequency in hertz (usually 800khz)
LED_FREQ_HZ = 800000

# DMA channel to use for generating signal (try 10)
LED_DMA = 3

# Set to 0 for darkest and 255 for brightest
# LED_BRIGHTNESS = 10
LED_BRIGHTNESS = 255
# LED_BRIGHTNESS = 0

# True to invert the signal (when using NPN transistor level shift)
LED_INVERT = False

# set to '1' for GPIOs 13, 19, 41, 45 or 53
LED_CHANNEL = 0

strip = Adafruit_NeoPixel(
  LED_COUNT,
  LED_PIN,
  LED_FREQ_HZ,
  LED_DMA,
  LED_INVERT,
  LED_BRIGHTNESS,
  LED_CHANNEL
  # LED_CHANNEL,
)

print("红色 LED 🔴")
strip.begin()

for led in range(0, LED_COUNT):
  strip.setPixelColor(led, Color(255, 0, 0))
  # sleep(1)

strip.show()

```

FoamyGuy, 2023-06-01:
@xgqfrms A few words of caution before anything else:

- In your power supply circuit it appears like you have 220v power connected in such a way where there are exposed bits of the wires making the connection with screw terminals. That seems potentially dangerous to me. If possible I would recommend try to find a way to connect that that is more secured and more difficult to accidentally touch. Or even better find fully enclosed power supply that outputs the 5v that you meed without the need for you to directly interact with the mains power.
-  Be careful running your strips of RGB LEDs with them all coiled up into a tight circle like that. It can generate considerable amounts of heat and potential damage the strip or start to melt the reel if it's wound onto one.

As for the different colors on your strip: What happens if you try setting the `bpp` value to `4` when you initialize the strip: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/main/neopixel.py#L110 like this:

```
pixels = neopixel.NeoPixel(board.D18, 60, bpp=4)
```

xgqfrms, 2023-06-01:
@FoamyGuy Thanks for your reminder.

> That's a 220V to 5V converter, so this should be safe.

input AC: 220V (+15%/-15%)
output DC: `5V/4A` ✅

FoamyGuy, 2023-06-01:
I meant that the way the wires are connected leaves them exposed so a person could accidentally touch them. Ideally that should be secured more so it's not possible to accidentally touch the live wires. 

What happens if you try like this:
```
pixels = neopixel.NeoPixel(board.D18, 60, order=neopixel.RGBW)
# ...

# then down inside of your loop:

    pixels[x] = (255, 0, 0, 0)  # <- add the fourth item in the tuple for the White in RGBW.
```

xgqfrms, 2023-06-01:
> These are the details of my RGB LED strip.

WS2812B RGB LED Strip
5050 LED
built-in IC
color RGB
0.3V/LED
60 LEDs/1M

[image]

https://detail.tmall.com/item.htm?id=672774081874

FoamyGuy, 2023-06-01:
Sorry, I don't really have any other ideas about what could be causing it to put different colors than the ones the code is setting. 

if you use `pixels.fill((255, 0, 0))` instead of looping and setting each pixel one at a time. Or if you use hex instead of tuples does it make any difference? like `pixels.fill(0xff0000)` I wouldn't expect either to change the behavior, but I'm out of any other ideas for troubleshooting.

xgqfrms, 2023-06-01:
## Buy a new LED strip❓

After trying many times, I still can't fix it.

Maybe it's a LED strip bug, who knows?

closed 2023-06-02 by xgqfrms

dhalbert, 2023-06-02:
@xgqfrms Did you fix it, or did you give up?

xgqfrms, 2023-06-11:
## 2023.06.12 update 

> It's not a bug of the `neopixel`.

```py
#!/usr/bin/env python3
# coding: utf8

import board
import neopixel
from time import sleep

# 60 LEDs
pixels = neopixel.NeoPixel(board.D18, 60)

def red_led_strip():
  for x in range(0, 60):
    pixels[x] = (255, 0, 0)
    sleep(0.1)
    # print("pixels[x] =", x, pixels[x])
    print("pixels[{}] =".format(x), x, pixels[x])

def clear_buffer():
  for x in range(0, 60):
    pixels[x] = (0, 0, 0)

try:
  red_led_strip()
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
except RuntimeError as error:
  print("error =", error, error.args[0])
  pass
except Exception as error:
  print("exception =", error)
  raise error
finally:
  print("after three seconds, auto clear buffer! 👻")
  sleep(3.0)
  # cleanup
  clear_buffer()
  print('clear 🚀')

"""
$ chmod +x ./led-strip.py
# ❌ Can't open /dev/mem: Permission denied
# $ ./led-strip.py

# ✅
$ sudo ./led-strip.py
"""
```

[image]

## The `BTF-LIGHTING` brand is a great product ✅

> This is an excellent product 🚀, I recommend buying it!

[image]

https://item.taobao.com/item.htm?id=522197656009

## The LED strip I bought earlier was a faulty product. ❌

> This is a shit product 💩, do not buy it anymore!

[image]

https://detail.tmall.com/item.htm?id=672774081874

xgqfrms, 2023-06-11:
# RGB version

```sh
#!/usr/bin/env python3
# coding: utf8

import board
import neopixel
from time import sleep

# 60 LEDs
pixels = neopixel.NeoPixel(board.D18, 60)

def red_led_strip():
  for x in range(0, 60):
    pixels[x] = (255, 0, 0)
    sleep(0.1)
    # print("pixels[x] =", x, pixels[x])
    # print("pixels[{}] =".format(x), x, pixels[x])
def green_led_strip():
  for x in range(0, 60):
    pixels[x] = (0, 255, 0)
    sleep(0.1)
    # print("pixels[x] =", x, pixels[x])
    # print("pixels[{}] =".format(x), x, pixels[x])
def blue_led_strip():
  for x in range(0, 60):
    pixels[x] = (0, 0, 255)
    sleep(0.1)
    # print("pixels[x] =", x, pixels[x])
    # print("pixels[{}] =".format(x), x, pixels[x])

def clear_buffer():
  for x in range(0, 60):
    pixels[x] = (0, 0, 0)

try:
  red_led_strip()
  sleep(1.0)
  green_led_strip()
  sleep(1.0)
  blue_led_strip()
  sleep(1.0)
  # while True:
  #   red_led_strip()
  #   sleep(1.0)
  #   green_led_strip()
  #   sleep(1.0)
  #   blue_led_strip()
  #   sleep(1.0)
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
except RuntimeError as error:
  print("error =", error, error.args[0])
  pass
except Exception as error:
  print("exception =", error)
  raise error
finally:
  print("after three seconds, auto clear buffer! 👻")
  sleep(3.0)
  # cleanup
  clear_buffer()
  print('clear 🚀')

"""
$ chmod +x ./led-strip.py
# ❌ Can't open /dev/mem: Permission denied
# $ ./led-strip.py

# ✅
$ sudo ./led-strip.py
"""
```

MasudRana72, 2023-06-19:
So, RPI 3B is the problem?
Do you fix the issue for 3B?

xgqfrms, 2023-06-19:
@MasudRana72  You can try it with the following hardware.

> Raspberry Pi `3B` + `12.5W ` official power supply ❓(I don't have this one power supply for now.)

```sh
$ vcgencmd get_throttled
```

## Raspberry Pi 12.5W `Micro USB` Power Supply

https://www.raspberrypi.com/products/micro-usb-power-supply/

https://datasheets.raspberrypi.com/power-supply/micro-usb-power-supply-product-brief.pdf

MasudRana72, 2023-06-19:
ok, at last the problem solved here for my RPI 3 b. I commented the line  [#dtparam](https://www.youtube.com/hashtag/dtparam)=audio=on     in /boot/config.txt file. 

it gives throttled=0x50000. and works in undervoltage also. Posting it might help someone. check this link(Pi 3B user): https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

xgqfrms, 2023-06-20:
# final solution ✅

```sh
$ vcgencmd get_throttled
throttled=0x0

```

```sh
$ sudo vim /boot/config.txt

$ sudo cat /boot/config.txt

# Enable audio (loads snd_bcm2835)
# ✅ fix: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/151
# dtparam=audio=on
```

```sh
$ sudo reboot

$ sudo ./pixel-strip-rgb.py
```

```py
#!/usr/bin/env python3
# coding: utf8

from time import sleep
from rpi_ws281x import PixelStrip, Color

# LED strip configuration:
LED_COUNT = 60        # Number of LED pixels.
LED_PIN = 18          # GPIO pin connected to the pixels (18 uses PWM!).
# LED_PIN = 10        # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating a signal (try 10)
# LED_BRIGHTNESS = 255   # Set to 0 for darkest and 255 for brightest
LED_BRIGHTNESS = 51   # Set to 0 for darkest and 255 for brightest
LED_INVERT = False     # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Create a NeoPixel object with the appropriate configuration.
strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)

# Initialize the library (must be called once before other functions).
strip.begin()
print("Test beginning ✅")

def clear_buffer():
  for i in range(LED_COUNT):
    strip.setPixelColor(i, Color(0, 0, 0))
  strip.show()
  sleep(1.0)

def rgb_test():
  # print("strip.numPixels() =", strip.numPixels())
  while True:
    for i in range(strip.numPixels()):
      # red led ✅
      strip.setPixelColor(i, Color(255, 0, 0))
      strip.show()
      sleep(0.1)
    sleep(1.0)
    for i in range(strip.numPixels()):
      # green led ✅
      strip.setPixelColor(i, Color(0, 255, 0))
      strip.show()
      sleep(0.1)
    sleep(1.0)
    for i in range(strip.numPixels()):
      # blue led ✅
      strip.setPixelColor(i, Color(0, 0, 255))
      strip.show()
      sleep(0.1)
    sleep(1.0)

try:
  rgb_test()
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
  clear_buffer()
  sleep(2.0)
except RuntimeError as error:
  print("error =", error, error.args[0])
  clear_buffer()
  pass
except Exception as error:
  print("exception =", error)
  clear_buffer()
  raise error
finally:
  print("clear buffer ✅")
  clear_buffer()
  sleep(3.0)
"""
$ sudo ./pixel-strip-rgb.py
"""
```