adafruit/Adafruit_CircuitPython_NeoPixel #163: LED Channel as one of parameters for NeoPixel

ExcitePrime, 2024-04-19:
Hello, can you please add the LED Channel as one of the parameters when creating an instance of NeoPixel? This is so that we could use GPIO13 and GPIO19, which are on a different LED channel
[image]

dhalbert, 2024-04-19:
This would be handled internally by https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/bcm283x/neopixel.py so you wouldn't need to specify the channel. 

But there are limitations that appear to make using both channelsnot possible. See https://github.com/adafruit/Adafruit_Blinka/issues/103, especially the last post.

closed 2024-04-19 by dhalbert

ExcitePrime, 2024-04-19:
> possible. See adafruit/Adafruit_Blinka#103

So the CircuitPython NeoPixel works only on PWM0 pins (GPIO12 and 18) and
not on PWM1 pins coz the LED_CHANNEL on the neopixel.py is explicitly set
to PWM0? That's too bad, I personally like the functions of the NeoPixel
class. I also essentially made a script that I just found out that is
similar to the multistrandtest.py (coz I'll be running two strips), so I
think I'll stick to rpi_ws281x for now. Thanks