adafruit/Adafruit_CircuitPython_NeoPixel #165: 16bit resolution per color?

stanelie, 2024-11-20:
Hello.

My application requires 16bit resolution per color channel. How do I set the bit depth?

The code examples show that the colors should be numbers between 0 and 255, but is this a hard restriction? Where is this set?

Thanks!

dhalbert, 2024-11-20:
The NeoPixels themselves physically only support 8-bit color resolution: they are sent 8-bit-each R, G, and B  values. So you can't display colors to a finer resolution than that.

stanelie, 2024-11-20:
Ahhh, this sucks. Back to the drawing board I guess.
Thanks for this precious info!

closed 2024-11-20 by dhalbert

stanelie, 2024-11-20:
Hello again.
I am currently working with the big 4 watt neopixel from here : https://www.adafruit.com/product/5163
After further investigation, I find that the onboard controller SM16714 has indeed 8 bit resolution per color, but on top of that, a 5 bit resolution current control _per color_ ! Combining both, this would maybe lead to more resolution then what is readily available.

"Use the 5-bit current control per color to lower the brightness without sacrificing resolution, and tune the current of individual colors, allowing for more accurate color mixing. "

Does the "brightness" parameter of the Neopixel library control this current globally? Would it be possible to control it for indivieual channels?

Reference : https://www.hida-led.com/sm16714-ic-componenets-product/

My next option will be to remove the chip and control the 4 LEDs directly with PWM signals from 4 discreet pins

dhalbert, 2024-11-20:
This driver doesn't support that. I guess if you don't supply the 20-bit current value, it still works. It's not clear who does support this extended protocol, except for the proprietary controllers. Some info here:
https://github.com/Aircoookie/WLED/pull/4221 (unmerged PR)
https://wled.discourse.group/t/sm16714-pixel-ic/9794

Are you using a whole string of these?

stanelie, 2024-11-20:
No, just one "pixel".
It's for theater lighting effects, and the resolution for the low levels is very important so you don't see the "steps".

dhalbert, 2024-11-20:
We don't have DMX controllers, but if you did, would DMX provide the resolution you need?

stanelie, 2024-11-20:
I am using Artnet, a network (UDP) version of DMX. This part is working.

On most light controller desks/software, you can configure a channel to use 16bit resolution. When set to this, a "channel" will encode it's level value on 2 adjacent DMX channels. The receiving device will use those 2 numbers to read a 16bit value number.

So now, I'm trying to have my Neopixel device behave like a proper 16bit output device.

jayceegee87, 2025-02-07:
Hi all,

Just chiming in that these are increasingly being utilized in outdoor LED applications (in my case LIFX string lights) given the ability to calibrate all of the individual color settings. It would be great to add compatibility and I happen to have one extra string available to fiddle around with if you need someone to check/test something. 

I also experienced the exact same artifacts as everyone else who has requested SM16714 support, but did identify an odd partial workaround at the expense of the overall brightness:  https://www.reddit.com/r/WLED/s/krDfmbn2co