adafruit/Adafruit_CircuitPython_NeoPixel #167: Neopixel not working on YD-RP2040

aguaviva, 2024-12-04:
I tried the following samples:

1- native sample
```
import board
import neopixel_write
import digitalio

pin = digitalio.DigitalInOut(board.NEOPIXEL)
pin.direction = digitalio.Direction.OUTPUT
pixel_off = bytearray([0, 0, 0])
neopixel_write.neopixel_write(pin, pixel_off)
```
2- installing neopixel using:
```
$ circup --path /media/raul/CIRCUITPY/ install neopixel
```
and running
```
# SPDX-FileCopyrightText: 2021 Kattni Rembor for Adafruit Industries
# SPDX-License-Identifier: MIT
"""CircuitPython status NeoPixel red, green, blue example."""
import time
import board
import neopixel

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1)

pixel.brightness = 0.3

while True:
    pixel.fill((255, 0, 0))
    time.sleep(0.5)
    pixel.fill((0, 255, 0))
    time.sleep(0.5)
    pixel.fill((0, 0, 255))
    time.sleep(0.5)
```

aguaviva, 2024-12-05:
it looks like there is a bridge I need to solder to make the neopixel work

closed 2024-12-05 by aguaviva

probonopd, 2024-12-18:
I am running into the same issue. Where did you need to solder a bridge? Thanks @aguaviva

probonopd, 2025-06-04:
R68 it is. Or is it "RGB"?

[image]